f4pga/f4pga-arch-defs #1: Create a netlistsvg skin with routing muxes and logic muxes

mithro, 2017-12-23:
Logic and routing muxes should look different in the diagrams generated.

mithro, 2018-03-07:
Vivado draws routing muxes like this;
[image]

And user controlled muxes like this;
[image]

Obviously the user controlled mux is a standard symbol, but the routing mux is not and a bit stupid. Lets **not** copy their symbol and instead come up with a better on.

mithro, 2018-03-07:
The difference between a **routing** and **logic** mux is if the **select signal** is controllable via logic or are place and route time. I think doing something which shows that is probably a good idea.

wifasoi, 2018-03-09:
What about an hatched pattern as a fill?
[image]
The only think need is create a pattern
```svg
    <pattern
       id="Hatch"
       width="4"
       height="4"
       patternTransform="rotate(-45 0 0)"
       patternUnits="userSpaceOnUse">
      <line
         x1="0"
         y1="0"
         x2="0"
         y2="4"
         style="stroke:black; stroke-width:1;fill:black;"/>
    </pattern>
```
and apply to the polygon with:
```svg
style=fill:url(#Hatch)
```
but i did not tested it yet

wifasoi, 2018-03-12:
@mithro if the symbol is ok, I'll precede to create a skin for netlistsvg; but I need more information, like a yosys netlist example with some routing muxes in it (or the name of the block, inputs, and output) and where put all my work.

mithro, 2018-03-12:
@daveshah1 Could you help @wifasoi with getting the needed information?

daveshah1, 2018-03-17:
From Yosys's POV both a routing mux and a logic mux are the same cell type (but a routing mux will be fed from a constant - parameters are resolved to a constant before elaboration, and the logic mux from an input). Any mux will be comprised of a cascade of `$mux` cells internally.

However, he `mux_gen.py` script, which can be assumed to be used for all muxes, will always instantiate a MUXn primitive inside a mux's simulation model. These build larger muxes out of `MUX2`. 

Probably the best way forward would be to create separate Verilog models inside the `vpr/muxes/logic` and a newly created `vpr/muxes/routing` folder for logic and routing muxes; and modify `mux_gen.py` to use these accordingly. Then netlistsvg skins could be created for all of these.

mithro, 2018-03-17:
Sounds like a good plan!