f4pga/f4pga-arch-defs #8: Generate pb_type.xml and model.xml from Verilog

mithro, 2017-12-23:
Currently you need to manually create the `pb_type.xml` and `model.xml` files needed for vpr.

These should in theory be able to.be generated from the `sim.v` files using Yosys to generate a .json file and then a Python script to generate the XML from the json output.

Only thing to worry about is how parameters are handled here.

Might need some type of naming convention for the logic verse routing muxes too.

mithro, 2018-02-18:
@daveshah1 is currently working on this.

mithro, 2018-02-18:
https://github.com/SymbiFlow/symbiflow-arch-defs/pull/37

daveshah1, 2018-02-25:
In the Artix-7 example, all parts of `CARRY4` except the mux (which still uses the `mux_gen.py` tool) and `W5FF` are now using this tool to generate their XML from their `sim.v` files.