f4pga/f4pga-arch-defs #14: Make xml_deps.py satsify pep8

mithro, 2017-12-26:
https://github.com/mithro/vpr-for-ice40/blob/master/utils/xml_deps.py

Cabalist, 2018-05-13:
This file no longer exists.  :)  Also `deps_xml.py` looks pretty good PEP8-wise.

mithro, 2018-05-13:
It has changed to [deps_xml as you mentioned](https://github.com/SymbiFlow/symbiflow-arch-defs/blob/master/utils/deps_xml.py).

If you think the file is all good, I guess we can close this issue?

Cabalist, 2018-05-13:
The only issue my linter throws is the "continuation indent" as referenced in #13.

Dhiraj240, 2019-03-26:
In reference to #13  please assign this also to me.Can i work on it ?

elmsfu, 2019-03-26:
@Dhiraj240 Feel free to work on this and create a PR. You can mention the issue ID in the PR description.

Dhiraj240, 2019-03-26:
Alright :+1:

Dhiraj240, 2019-03-27:
@elmsfu Please close this issue it is already satisfying pep8.

elmsfu, 2019-03-28:
`yapf` has no issues with this file.

closed 2019-03-28 by elmsfu