f4pga/f4pga-arch-defs #55: Change to using `specify` blocks for providing timing information.

mithro, 2018-02-25:
Currently we are using Verilog attributes for timing information because Yosys doesn't support `specify` blocks. `specify` blocks are actually the proper way to provide timing information in Verilog, see below;

From http://verilog.renerta.com/source/vrg00044.htm;
> #### Example 1
> ```verilog
> module ...
> ...
> specify
>   (In => Out) = (10);
> endspecify
> ...
> endmodule
> ```
> A specify block with only a path declaration. Delay between input In and output Out is 10 time units.
> 
> #### Example 2
> ```verilog
> module ...
> ...
> specify
>   specparam TRise = 10,
>   TFall = 15;
>   (In => Out) = (TRise, TFall) ;
> endspecify
> ...
> endmodule
> ```
> Specparam declaration with two parameters TRise and TFall to specify delays on rising transition and falling transition.
> 
> #### Example 3
> ```verilog
> module ...
> ...
> specify
>   specparam TRise = 10,
>   TFall = 15;
>   (In => Out) = (TRise, TFall) ;
>   $setup(Data_in, posedge Clock, TRise) ;
> endspecify
> ...
> endmodule
> ```

daveshah1, 2018-02-25:
This ultimately depends on support for these in a parser/elaborator that gives us useful output. At the moment Yosys isn't even able to ignore them, let alone handle them (see [https://github.com/YosysHQ/yosys/issues/506](Yosys #506)). 

I think Icarus Verilog is the only open source tool at the moment that supports them, but I don't know if we can get any output from that in a useful form. Maybe it would be easier and cleaner to add support for them in Yosys (e.g. so it dumped them as attributes or similar in the JSON) in the long run though.

udif, 2018-03-04:
Please take a look at https://github.com/YosysHQ/yosys/pull/510 as a starting point.
(Parsing done, but no AST created).

mithro, 2018-03-16:
Verilog To Routing supports [Post Implementation Timing Simulation](http://docs.verilogtorouting.org/en/latest/tutorials/timing_simulation/?highlight=timing) by generating a "Standard Delay Fromat (SDF)" file. Maybe we should use that file format instead?