f4pga/f4pga-arch-defs #59: Figure out how to write unit tests for the sim.v files.

mithro, 2018-03-02:
We want a structured way to write unit tests for the sim.v files (and a way to run them).

We want the unit tests to be compatible with at a minimum;
 * Yosys
 * Verilator
 * Icarus Verilog

But making it compatible with commercial simulators would be good too.

I *think* we want to use [Cocotb](http://cocotb.readthedocs.io/en/latest/introduction.html) but I have never used it myself.

mcmasterg, 2018-03-02:
I don't have a whole lot to say on this / no strong opinion.

But I can add:
-UVM is way overkill for this type of application. Don't even consider it
-For Skybox I rolled my own framework that loosely resembled JUnit
-I've never used Cocotb. I'd need to try it out before having an opinion on it. I'm up to exploring it / seeing a demo

mithro, 2018-03-02:
@cliffordwolf - Do you have an opinion on how we should be writing unit tests for these models? I'm guessing you are going to say we should be using formal verification? :-P

mithro, 2018-03-02:
@awygle - FYI.

elmsfu, 2018-03-14:
I've started to look at demonstrating a simple test using [cocotb](http://cocotb.readthedocs.io/en/latest/introduction.html) and [VUnit](http://http://vunit.github.io/).

I have a [branch](https://github.com/elmsfu/symbiflow-arch-defs/tree/cocotb_tests) with a simple test using cocotb (think I found an bug in mux2.sim.v).

VUnit doesn't currently support Icarus Verilog, but people are working on it (see vunit/vunit#188). I haven't had success running this branch yet (getting a segfault during compilation).

_Other thoughts:_

VUnit is a good choice to drive tesbtenches on an existing codebase. Each testbench needs a few lines added to integrate with VUnit. Running VUnit the output is very similar to many unit test frameworks. It looks like it can be extended to report details to other processes (eg [octave](https://github.com/LarsAsplund/matlab_integration)) and do additional checks.

cocotb takes a different approach and borrows from UVM. The test cases are written in python which opens all the flexibility of python to how to drive DUTs. It also has a built-in concept for generating tests, but I haven't looked at that yet.

Both support GHDL and integration with Jenkins (which hopefully means easily with travis-ci).

elmsfu, 2018-03-22:
Yes.
> Cocotb encourages the same philosophy of design re-use and randomised testing as UVM [[link](http://cocotb.readthedocs.io/en/latest/introduction.html#how-is-cocotb-different)]

They also use the _scoreboard_, _driver_, and _monitor_ patterns which I believe is terminology shared with/borrowed from UVM. These are available but not required.