f4pga/f4pga-arch-defs #66: Figure out why echo doesn't support '-e' on travis

mithro, 2018-03-07:
On Travis we are getting `-e <string>` when we use `echo -e <string>`. On my local computer we just get `<string>` but echo is correctly interpreting the escape characters.

From the man page;
> `-e     enable interpretation of backslash escapes`

I'm guessing it has something to do with shell builtins verse real commands or something?

mithro, 2018-03-07:
Example;
https://travis-ci.org/mithro/symbiflow-arch-defs/builds/350461107

[image]

elmsfu, 2018-03-09:
I think this is because make uses `sh` and the builtin `echo` doesn't support that option. https://stackoverflow.com/questions/589276/how-can-i-use-bash-syntax-in-makefile-targets

A few solutions:
1. You can just not use `-e` as by default sh respects the escape characters
1. set `SHELL = bash` to force using bash
1. define ECHO based on `SHELL` value and replace `echo -e` with `$(ECHO)`

Defining ECHO probably provides the most flexibility.

mithro, 2018-03-09:
https://travis-ci.org/SymbiFlow/symbiflow-arch-defs/builds/351175350

PR #76 (2018-03-09): Using bash as shell in Makefile
Hopefully fixes #66.
closed 2018-03-09 by mithro