f4pga/f4pga-arch-defs #69: Make sure all dependencies are documented and available in the conda environment

mithro, 2018-03-07:


mithro, 2019-05-07:
@litghost I think this is done?

litghost, 2019-05-07:
This issue never described the problem, so I have no idea.

mithro, 2019-05-08:
It's all in the title
> Make sure all dependencies are documented and available in the conda environment

litghost, 2019-05-08:
That is vague.  For example, not all dependencies for SymbiYosys are present (e.g. #572 ).  I stand by my statement that this issue is ill formed, and I don't really know what exactly success is for it.

mithro, 2019-05-08:
> Dependencies required to run the default set of make targets can be automatically provided by the conda environment and `make env` command.
> 
> These requirements are also written down in a document somewhere to enable people who do not wish to use the conda environment to provide them manually. This should probably include the environment variables required to be set.

@litghost Is that better?

litghost, 2019-05-08:
I'd say https://github.com/SymbiFlow/symbiflow-arch-defs/issues/564 indicates there is still work here, but besides #564, I'm not aware of what is left.