f4pga/f4pga-arch-defs #71: Figure out why make takes ~2 seconds to start

mithro, 2018-03-08:
Currently when you type `make` it takes about 2 seconds to start running.

I believe this is because of the include system that is now used for getting proper dependency info. I believe this because when running targets which don't use the include system (such as `make clean`) the make command is almost instant.

Make is most certainly slow because I'm doing something silly in when doing the includes. Searching all for all the files (IE the utils/listfiles.py function) is extremely fast.

elmsfu, 2018-03-30:
I noticed that [this line](https://github.com/SymbiFlow/symbiflow-arch-defs/blob/master/make/deps.mk#L46) is a tautology because $(FROM) is by default empty. But I don't understand the intention of this section. Calling find_files for every dependency seems expensive for a conditional.

mithro, 2018-03-31:
That does look like a bug. I think it should probably be `ifneq (,$$(call find_files,$$(ADD_DEP_FROM)))`?The question is why are we not getting the warning message then?

closed 2018-08-22 by litghost