f4pga/f4pga-arch-defs #72: Make it so you don't need to create the redirection makefiles to generate .git/info/exclude file

mithro, 2018-03-08:
Currently to work out if we want to ignore a Makefile we do the following;
```make
redir:
	@for DIR in $$($(UTILS_DIR)/listdirs.py); do \
	  if [ ! -e $$DIR/Makefile ]; then \
	    ln -sf $(abspath $(TOP_DIR)/make/redir.mk) $$DIR/Makefile; \
	    export NEW_MAKEFILE=$$(echo $$DIR/Makefile | sed -e's@^$(TOP_DIR)/@@'); \
	    echo "Creating redirect makefile '$$NEW_MAKEFILE'"; \
	  fi; \
	done

.git/info/exclude.redir: $(TOP_DIR)/make/gen.mk | redir
	$(file >$(TARGET),$(call gitexclude_comment,Makefile redirect))
	@for DIR in $$($(UTILS_DIR)/listdirs.py); do \
	  if [ $$(python -c"import os.path; print(os.path.realpath('$$DIR/Makefile'))") = $(abspath $(TOP_DIR)/make/redir.mk) ]; then \
	    export NEW_MAKEFILE=$$(echo $$DIR/Makefile | sed -e's@^$(TOP_DIR)/@@'); \
	    echo  "$$NEW_MAKEFILE" >> $(TARGET); \
	  fi; \
	done
```

This means that the redirection Makefiles *have* to exist before we can generate `.git/info/exclude.redir`

This is the only real place we need to *actually* create something for the make system. In all other cases things work with just the **potential** that a file **could** exist.

elmsfu, 2018-03-30:
If I understand the current [master](https://github.com/SymbiFlow/symbiflow-arch-defs/blob/854feee1fe1d9bdaca916427b18a746ab4886aa4/make/gen.mk#L18) it looks like a work around is in place. Is it still generating them before generating the exclude.redir?

mithro, 2018-03-31:
I think so? I'll check tomorrow...

closed 2018-08-22 by litghost