f4pga/f4pga-arch-defs #92: Port the iCE40 CLB to Verilog and generate the pb_type from it

mithro, 2018-04-20:
@daveshah1 [created a tool for generating Verilog to Routing XML files from verilog](https://github.com/SymbiFlow/symbiflow-arch-defs/tree/master/utils/vlog). 

The goal is to eventually have fully simulatable Verilog descriptions of all the *actual* hardware (as opposed to the verilog macros for instantiating hardware). We have been working on this for the Artix 7 and we should do the same for the iCE40 hardware.

mithro, 2018-04-20:
@jhol this isn't a high priority at the moment but something we should look at eventually.

daveshah1, 2018-04-21:
I think a few more features, in particular support for generating `num_pb` from Verilog may be needed for this to work, but it should be doable.

jhol, 2018-04-23:
This is an interesting concept - the idea of being able to throw arbitrary FPGA Verilog into VPR sounds quite appealing. But also I get a strong sense of YAGNI/KISS about it.

It would be cool if you could make it work, but I'm not convinced that it's really going to work as seamlessly as we need, because there seems to be a bit of an impedance mismatch between the information present in Verilog, and the inputs to VPR.

The arch XML format contains 4 key things:

1. List of BLIF models
2. Description of the pb_types
3. Routing description
4. Tile layout

Verilog is only going to help with 1 and part of 2.

There will need to be a lot of additional XML included in for the rest of the information.

For 2, a lot of the "art" of architecture XML, is generating a structure that sidesteps VPR's various quirks. It will be quite hard to make a general purpose Verilog to pb_type conversion step. I have a feeling that there will be a large number of edge cases which have to be supported in principle, but it's impossible to know whether that code will pay off until it is tested with a wider selection of FPGAs. The alternative is a half-arsed conversion algorithm, but this will simply break for any new FPGAs we add.

I don't think it's possible to properly describe the requirements of a Verilog to pb_type converter at this time.

As you know, I advocate simply translating the XML+XSLT code into Python. The main issue with the arch XML is the repetitious nature of the `pb_type` and routing description XML. Whatever approach we take, it should aim to boost the "information density" of the architecture description - like a compression algorithm.

The true description of the ice40 internal structure could be described in words and pseudo-code in just a few pages of paper, so the format we choose to use should roughly equate to this.

As I showed in this proof of concept...

https://github.com/jhol/icestorm/blob/vtr-py/icebox/vtr.py

...just converting the XML to Python yields great gains in information density. Simply being able to wrap all repetitions up in `for` loops and functions, means the code never states a piece of information more than once.

This code makes an attempt at all 4 of the things described above, and comes out at less than 400-LOC, including comments.

In my opinion, the way forward is to take the XML+XSLT code and translate it into any popular scripting language. For the iCE40, this would take at most 2-days - probably could be done in 1-day. Then once the conversion is complete, any common patterns between iCE40 and 7-series can be factored out into a framework library. Then over time, as support more FPGAs is added, even more common patterns will emerge, and the framework can become get more featureful.

Once a decent number of FPGAs is supported there may be an appetite for further improving the information density by converting parts of the description into a domain-specific language - that language could be Verilog, or some other pre-existing or bespoke language might be more suitable.

The nice thing about what I'm laying out is that the task can be done in easy incremental stages without committing to implement a complex solution before the requirements are fully understood.

Any thoughts? Agree? Disagree?