f4pga/f4pga-arch-defs #99: Create a Python library for generating VtR arch.xml file

mithro, 2018-04-26:
Currently we generate all the arch.xml files by either;
 * Hand using XML includes
 * @davesha's Verilog to arch.xml generator.

As @jhol points out, it would be much easier to generate a lot of the definitions in Python and then dump the XML from that.

This would also make it easier to do the many variations of the testing architectures.

mithro, 2018-04-26:
We could also look at [migen](https://github.com/m-labs/migen) to generate Verilog from the  same structures.

mithro, 2018-04-26:
The Verilog to arch generator could use this API too...

mithro, 2018-04-30:
@jhol has some examples of how this might work here -> https://github.com/jhol/icestorm/blob/vtr-py/icebox/vtr.py

mridubhatnagar, 2018-06-22:
Have worked on create an XML, parsing an XML, with Python.

Is this what the issue needs?

mithro, 2018-06-22:
Hi @mridubhatnagar,

We want to be able to programmatically generate Verilog to Routing [FPGA architecture files](http://docs.verilogtorouting.org/en/latest/arch/reference/). These XML files describe the internals of an FPGA.

We could just use the DOM to do this, but the DOM is not a great Python API. It also sucks if we change from using XML format to some other type of format.

So what we are after is a Python API for creating these FPGA structures and then a way to convert these FPGA structures to and from the XML format.

You can see how @jhol was doing this [here](https://github.com/jhol/icestorm/blob/vtr-py/icebox/vtr.py) and use that as inspiration. You can also see our [rr_graph library](https://github.com/SymbiFlow/symbiflow-arch-defs/blob/master/utils/lib/rr_graph/graph.py) as inspiration too.

One approach to this would be creating an XML schema for the file format and then auto-generating the Python API from that, see the following examples;
 * http://www.davekuhlman.org/generateDS.html
 * http://pyxb.sourceforge.net/
 * https://github.com/knorby/pyxsd

[lxml.de's objectify](http://lxml.de/objectify.html) might be another way to do this.

The first goal would be to easily generate the same architecture files as found in the [testarch](https://github.com/SymbiFlow/symbiflow-arch-defs/tree/master/testarch/devices) repository before building up towards being able to generate much more complicated structure.

If you want to chat more, find mithro on the [##openfpga IRC channel](http://webchat.freenode.net/?channels=#openfpga) or [email me at me@mith.ro](mailto:me@mith.ro).

Hope that helps!

mithro, 2018-06-22:
Also see the related issue here -> https://github.com/SymbiFlow/ideas/issues/4

mithro, 2019-04-08:
There is a WIP at https://github.com/leon575777642/vprgen/tree/master/vprgen