f4pga/f4pga-arch-defs #101: Refactor build system

elmsfu, 2018-04-27:
Main goals would be to speed up and simplify.

Python is already used to detect dependencies, instead of generating a lot of files in .deps it could just write out the dependencies and build rules from python. I think this would remove a lot of redirection and speed up build time.

Another idea would be to put all generated files in a separate results directory to remove the need of exclude generation.

mithro, 2018-06-06:
Options to consider;
 * Custom Python scripts which just generate Makefiles
 * Custom Python scripts which [generate Ninja](https://ninja-build.org/)
 * [meson](https://mesonbuild.com/)
 * [waf](https://github.com/waf-project/waf)

mithro, 2018-06-06:
I attempted to create a meson build script here -> https://github.com/mithro/symbiflow-arch-defs/blob/meson/meson.build

mithro, 2018-06-06:
I /really/ want to use waf -- but my spider sense is saying that we will regret that choice 6 months from now.

mithro, 2018-06-06:
I think we should also consider `cmake` as it is used by Verilog-To-Routing and the prjxray -- but I really dislike it...

mithro, 2018-06-06:
We should also consider using [`conda environments`](https://conda.io/docs/user-guide/tasks/manage-environments.html#building-identical-conda-environments) for doing the env / conda stuff too...

closed 2018-08-22 by litghost