f4pga/f4pga-arch-defs #102: Pylint (or similar) on all python code as part of CI

elmsfu, 2018-04-27:
* Agree on pylint configuration
* Bring python into compliance
* Then run pylint on all file as part of travis.

elmsfu, 2018-04-30:
Hadn't heard of yapf. Looks good for formatting. Do you like the idea of adding pylintrc like I started in #97. I also started a [branch](https://github.com/elmsfu/symbiflow-arch-defs/tree/pylint_cleanup) to start making python code comply.

mithro, 2018-08-09:
FYI - the prjxray repo uses this -> https://github.com/SymbiFlow/prjxray/blob/master/.style.yapf

elmsfu, 2019-03-15:
Found an old note that I used the command below to report git-tracked files that would have changes if yapf was run on it.
```
for ii in `git ls-tree HEAD --name-only -r | grep \.py$`; do ./build/env/conda/bin/yapf -d $ii > /dev/null; if [[ $? != 0 ]]; then echo $ii; fi; done
```

elmsfu, 2019-04-05:
Started adding pyflakes [here](https://github.com/elmsfu/symbiflow-arch-defs/tree/add_pyflakes). Will need to address lots of compliance.