f4pga/f4pga-arch-defs #121: Figure out why vpr can't route between PLB.lutff_0_out[0] and BLK_TL-PIO_T.io_0_D_OUT[0]

mithro, 2018-05-14:
Figure out why vpr can't route between `PLB.lutff_0_out[0]` and `BLK_TL-PIO_T.io_0_D_OUT[0]`

```
Cannot route from BLK_TL-PLB.lutff_0_out[0] (rr_node: 3352 type: SOURCE ptc: 35 xlow: 5 ylow: 5) to BLK_TL-PIO_T.io_0_D_OUT[0] (rr_node: 5620 type: SINK ptc: 0 xlow: 7 ylow: 10) to -- no possible path
Warning 2290: Unable to route between blocks at (5,5) and (7,10) to characterize delay (setting to inf)
Warning 2291: Empty heap occurred in get_heap_head.
Warning 2292: Some blocks are impossible to connect in this architecture.
Cannot route from BLK_TL-PLB.lutff_0_out[0] (rr_node: 3352 type: SOURCE ptc: 35 xlow: 5 ylow: 5) to BLK_TL-PIO_R.io_0_D_OUT[0] (rr_node: 5732 type: SINK ptc: 0 xlow: 8 ylow: 5) to -- no possible path
Warning 2293: Unable to route between blocks at (5,5) and (8,5) to characterize delay (setting to inf)
Warning 2294: Empty heap occurred in get_heap_head.
Warning 2295: Some blocks are impossible to connect in this architecture.
Cannot route from BLK_TL-PLB.lutff_0_out[0] (rr_node: 3352 type: SOURCE ptc: 35 xlow: 5 ylow: 5) to BLK_TL-PIO_R.io_0_D_OUT[0] (rr_node: 5760 type: SINK ptc: 0 xlow: 8 ylow: 6) to -- no possible path
Warning 2296: Unable to route between blocks at (5,5) and (8,6) to characterize delay (setting to inf)
Warning 2297: Empty heap occurred in get_heap_head.
Warning 2298: Some blocks are impossible to connect in this architecture.
Cannot route from BLK_TL-PLB.lutff_0_out[0] (rr_node: 3352 type: SOURCE ptc: 35 xlow: 5 ylow: 5) to BLK_TL-PIO_R.io_0_D_OUT[0] (rr_node: 5788 type: SINK ptc: 0 xlow: 8 ylow: 7) to -- no possible path
Warning 2299: Unable to route between blocks at (5,5) and (8,7) to characterize delay (setting to inf)
Warning 2300: Empty heap occurred in get_heap_head.
Warning 2301: Some blocks are impossible to connect in this architecture.
Cannot route from BLK_TL-PLB.lutff_0_out[0] (rr_node: 3352 type: SOURCE ptc: 35 xlow: 5 ylow: 5) to BLK_TL-PIO_R.io_0_D_OUT[0] (rr_node: 5816 type: SINK ptc: 0 xlow: 8 ylow: 8) to -- no possible path
Warning 2302: Unable to route between blocks at (5,5) and (8,8) to characterize delay (setting to inf)
Warning 2303: Empty heap occurred in get_heap_head.
Warning 2304: Some blocks are impossible to connect in this architecture.
Cannot route from BLK_TL-PLB.lutff_0_out[0] (rr_node: 3352 type: SOURCE ptc: 35 xlow: 5 ylow: 5) to BLK_TL-PIO_R.io_0_D_OUT[0] (rr_node: 5844 type: SINK ptc: 0 xlow: 8 ylow: 9) to -- no possible path
Warning 2305: Unable to route between blocks at (5,5) and (8,9) to characterize delay (setting to inf)
Placement delay look-up took 0.311494 seconds
```

mcmasterg, 2018-06-11:
What is an easy way to test this?

mithro, 2018-06-11:
Run vpr with the rr_graph? `make route` on the 4mcmaster branch for example. Also can be tested with https://github.com/SymbiFlow/symbiflow-arch-defs/pull/133