f4pga/f4pga-arch-defs #131: Investigate weird routing in ice40 arch

mithro, 2018-05-26:
Taking a look at this picture there are two weird things;
[image]

The output pin of this BLK_TL-PLB seems to be connected to a span12 in an adjacent column. Highlighted below;
[image]

The local tracks seem to be connected to an adjacent column.
Highlighted below;
[image]

mcmasterg, 2018-06-11:
To be clear, this ticket is about concerns over the routing fabric, not the routing on the fabric?

Your main concerns are that the spans are not located closer to what they are connecting? Do you believe this is wrong based on some documentation? Or just find it suspicious that they aren't adjacent?

mithro, 2018-06-11:
Why are these connected to columns adjacent? It seems suspicious.

mithro, 2018-07-03:
This seems to be fixed by https://github.com/SymbiFlow/symbiflow-arch-defs/pull/150

closed 2018-07-03 by mithro