f4pga/f4pga-arch-defs #144: Create an ice40 test for the packing of different types of flip flops together

mithro, 2018-06-06:
From http://www.clifford.at/icestorm/logic_tile.html

> Clock, clock enable, and set/reset inputs are shared along the 8 logic cells. So is the bit that configures positive/negative edge for the flip flops. But the three configuration bits that specify if the flip flop should be used, if it is set or reset by the set/reset input, and if the set/reset is synchronous or asynchronous exist for each logic cell individually.

There also seem to be the following SB_FF models;
```xml
  <model name="SB_DFF">
  <model name="SB_DFFE">
  <model name="SB_DFFER">
  <model name="SB_DFFES">
  <model name="SB_DFFESR">
  <model name="SB_DFFESS">
  <model name="SB_DFFR">
  <model name="SB_DFFS">
  <model name="SB_DFFSR">
  <model name="SB_DFFSS">
```

This means you can only pack certain types of flip flops together into a tile. Need a test which explicitly looks for this.

mithro, 2018-06-15:
There is one added here -> https://github.com/mithro/symbiflow-arch-defs/tree/4mcmaster/tests/ice40/ffpack

This needs https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/365 to be merged into VPR before it works.

elmsfu, 2018-07-25:
After some discussion, it seems like a good approach is to generate (with python) many possible strings of differing FF types and then use formal verification to compare the original to generated verilog from the asc file.

We think there are 20 types of FF

negclk | SR | async | set_norest | cen | type name
----------|------|---------|----------------|------|---------------
0|0|X|X|0| DFF
0|0|X|X|1| DFFE
0|1|0|0|0| DFFSR
0|1|0|0|1| DFFESR
0|1|0|1|0| DFFSS
0|1|0|1|1| DFFESS
0|1|1|0|0| DFFR
0|1|1|0|1| DFFER
0|1|1|1|0| DFFS
0|1|1|1|1| DFFES
1|0|X|X|0| DFFN
1|0|X|X|1| DFFNE
1|1|0|0|0| DFFNSR
1|1|0|0|1| DFFNESR
1|1|0|1|0| DFFNSS
1|1|0|1|1| DFFNESS
1|1|1|0|0| DFFNR
1|1|1|0|1| DFFNER
1|1|1|1|0| DFFNS
1|1|1|1|1| DFFNES