f4pga/f4pga-arch-defs #149: Create a tool to extract the library definitions from Xilinx PDF documents

mithro, 2018-06-21:
To be compatible with code written specifically for Xilinx Artix-7 parts we need a file which describes the Verilog and VHDL interfaces uses on those parts. These definitions can be found in the [UG953 Vivado 7 Series libraries - Xilinx PDF Document](https://www.xilinx.com/support/documentation/sw_manuals/xilinx2014_2/ug953-vivado-7series-libraries.pdf).

We need to extract **only** the minimum amount of information required to be compatible. This list is currently the following information;
 * The module name.
 * The port name.
 * The size and direction of the ports.
 * The order of the ports.

There is a [start of such an extractor here](https://github.com/SymbiFlow/symbiflow-arch-defs/tree/master/artix7/libraries).

mjasperse, 2019-02-13:
I've implemented a parser that successfully extracts the module ports information directly from the UG953 PDF using the pdfminer package. I think the direct approach is more reliable than an intermediary pdf-to-text converter because there's a lot of ambiguity in how text gets serialised. I avoid this by using the raw positioning data for all text items in each "Port Descriptions" table. Module names are read directly from the ToC, and inconsistencies corrected in post-processing.

What's your preferred workflow? Fork and branch, add to xc7/libraries and submit a PR?

mithro, 2019-02-13:
@mjasperse - Fork add to xc7/libraries and submit a pull request.

Do make sure that your commits contain a [DCO line](https://github.com/wking/signed-off-by) too.

elmsfu, 2019-03-19:
This looks fixed in #374. Reopen if I'm incorrect.

closed 2019-03-19 by elmsfu