f4pga/f4pga-arch-defs #159: Generate the iCE40 routing graph rather than importing from icebox

mithro, 2018-07-03:
Currently the way to get a real rr_graph is to use the [ice40/utils/ice40_import_routing_from_icebox.py](https://github.com/mithro/symbiflow-arch-defs/blob/4mcmaster/ice40/utils/ice40_import_routing_from_icebox.py) script.

However, it would be much nicer if we could generate the routing without needing icebox. The script at [ice40/utils/ice40_generate_routing.py](https://github.com/mithro/symbiflow-arch-defs/blob/4mcmaster/ice40/utils/ice40_generate_routing.py) starts doing that.

mithro, 2018-07-03:
This is related to https://github.com/verilog-to-routing/vtr-verilog-to-routing/issues/286

mithro, 2018-07-03:
I'm working on this here -> https://github.com/mithro/symbiflow-arch-defs/tree/ice40-generate

mithro, 2018-07-15:
This is being worked on in https://github.com/SymbiFlow/symbiflow-arch-defs/pull/164

tmichalak, 2019-01-04:
I started looking into the problem. However, since the last commit in the related PR is six months old, I rebased the fixes from @mithro on the current master first and then ran the flow. The ice40_generate_routing.py script aborted with invalid key error due to the fact that the "global" segment in the virtual rr_graph was missing. I noticed that there is "tile_global" segment in the virtual rr_graph though. I am not sure if the "global" segment has been replaced with "tile_global", but tried to continue the flow by changing the ice40_generate_routing (https://github.com/antmicro/symbiflow-arch-defs/commits/159_ice40_generate). Unfortunately, vpr failed during routing with: 
``
Cannot route from BLK_TL-PLB.lutff_7/out[0] (rr_node: 3184 type: SOURCE ptc: 42 xlow: 4 ylow: 13) to BLK_TL-PLB.lutff_0/in[1] (rr_node: 6538 type: SINK ptc: 1 xlow: 7 ylow: 3) to -- no possible path
Failed to route connection from '$0\counter[28:0][0]' to '$abc$1427$n63' for net 'counter[0]'
Routing failed.
Routing took 0.35 seconds
Circuit is unroutable with a channel width factor of 100.
``
I am not sure if tracing what keeps the routing from finishing is the way to go. Is my assumption regarding the "global" segment being replaced by the "tile_global" correct in the first place?