f4pga/f4pga-arch-defs #181: Write a Python library for decoding Artix-7 wire names

mithro, 2018-07-11:
The names of wires in the series-7 FPGA have meaning, for example `SW4BEG0` might mean 'the beginning of a wire which runs south west 4 tiles`.

There is the start of some code here -> https://github.com/SymbiFlow/symbiflow-arch-defs/blob/474d3ed44647c1dc53eafddb16ba9eb0e025506a/artix7/utils/prjxray-routing-import.py#L308-L343

The [Project X-Ray `tileconn.json`](https://github.com/SymbiFlow/prjxray-db/blob/master/artix7/tileconn.json) has a list of names we know.

But it would nice to have a proper library with tests and such.

mithro, 2019-01-31:
@litghost Does this library exist somewhere?

litghost, 2019-01-31:
We don't need it?

litghost, 2019-01-31:
Also, in special cases, the interpretation is wrong.  At the top and bottom of every INT tile are turn around blocks.  I believe there are similiar constructs above large blocks, like PCI

closed 2019-01-31 by litghost

BalasubramanyamEvani, 2019-05-17:
Hi ! is this issue already assigned or still open to work on. I am new to the project and would  like to contribute. Can anybody guide me through this ? 
Thank you.

mithro, 2019-05-17:
@BalasubramanyamEvani This idea is still open to being done!

First step is to probably write up a short document which has a [list of the wires from the FASM prjxray database](https://github.com/SymbiFlow/prjxray-db/blob/master/artix7/tileconn.json) and figure out what they mean in "human terms". Once that is done, it should be pretty easy to start writing some Python code to extract this information....

BalasubramanyamEvani, 2019-05-17:
@mithro Ok got it ! I will start working upon it then. I will first go through the json file and find respective meanings of the wires, the work on the python script.

bangpradyumna, 2020-03-27:
@mithro Is this issue still relevant and if yes, Can I please work on it ?

Thanks

mithro, 2020-03-27:
Yes, this is still a relevant issue and anyone can work on it. (If you are thinking in terms of GSoC, it probably isn't big enough to be a GSoC project by itself.)