f4pga/f4pga-arch-defs #188: tinyfpga b2 wrong fpga

mcmasterg, 2018-07-14:
in ice40k.mk

```
# TinyFPGA B2
# iCE40-LP8K-CM81
# ---------------------------------------------
ifeq ($(BOARD),tinyfpga-b2)
DEVICE=hx8k
PACKAGE=cm81
```

which is clearly wrong. Tim says that there might be an outstanding PR to fix this. Placeholder ticket to make sure this gets taken care of

mcmasterg, 2018-07-14:
also note

```
ls ~/symbiflow-arch-defs/ice40/devices/layouts/icebox/ |grep cm81.pinmap.csv
hx1k.cm81.pinmap.csv
hx8k.cm81.pinmap.csv
lp1k.cm81.pinmap.csv
```

Maybe this will get fixed if the makefile is mixed?

mithro, 2018-07-14:
Hrm...
```
../../../utils/ice40_list_layout_in_icebox.py  | grep cm81
lp1k.cm81
hx1k.cm81
hx8k.cm81
```

mithro, 2018-07-14:
From icetime;
```python
for db in "lp384 lp1k lp8k hx1k hx8k up5k".split():
```

mithro, 2018-07-14:
```
# utils/ice40_list_layout_in_icebox.py  | grep cm81
lp1k.cm81
hx1k.cm81
lp8k.cm81
hx8k.cm81
```

IceBox doesn't seem to have a way to tell the `hx8k.cm81` doesn't exist.

commit e83db3f1 (2018-07-14): ice40: Fix tinyfpga-b2 part name.

Fixes #188.
closed 2018-07-14 by mithro