f4pga/f4pga-arch-defs #209: Create a "pretty printer" for the VTR .net file format

mithro, 2018-07-26:
The VTR [packed XML `.net` is hard to read](https://vtr-verilog-to-routing.readthedocs.io/en/latest/vpr/file_formats.html#packed-netlist-format-net).

It would be good if we had a Python based tool that printed more human readable output. You can take inspiration from how [utils/print_graph.py](utils/print_graph.py) tries to make the `rr_graph.xml` format easier to read.

jeffherman, 2018-08-01:
This seems pretty straight forward. Do you have a collection of `.net` files aside from the one in the snippet in the doc?

mithro, 2018-08-02:
We don't really have a huge set of samples - the repo can generate them. There is also an XML schema file here -> https://github.com/SymbiFlow/symbiflow-arch-defs/blob/master/common/xml/packed_netlist.xsd

```xml
<?xml version="1.0"?>
<block name="example.net" instance="FPGA_packed_netlist[0]" architecture_id="SHA256:17c4520f123fca600b54f19bee352b4e4f5fb2479732a207449b8fcde86f73b5" atom_netlist_id="SHA256:ab08d1d797cab5dcda3826ae6ccaed7c34b2f49e0aedaf3c33c8b3dcc89e1385">
	<inputs>clk cen rst ina inb</inputs>
	<outputs>out:outa out:outb out:outc out:outd out:vec0[0] out:vec0[1] out:vec0[2] out:vec0[3] out:vec0[4] out:vec1[0] out:vec1[1] out:vec1[2] out:vec1[3] out:vec1[4]</outputs>
	<clocks>clk</clocks>
	<block name="out:vec0[0]" instance="BLK_TL-PIO[0]" mode="PAD_IS_OUTPUT">
		<inputs>
			<port name="D_OUT">vec0[0] open</port>
			<port name="OUT_ENB">open</port>
			<port name="CEN">open</port>
			<port name="LATCH">open</port>
		</inputs>
		<outputs>
			<port name="D_IN">open open</port>
			<port name="PACKAGE_PIN">open</port>
		</outputs>
		<clocks>
			<port name="INCLK">open</port>
			<port name="OUTCLK">open</port>
		</clocks>
		<block name="out:vec0[0]" instance="PAD[0]" mode="default">
			<inputs>
				<port name="D_OUT">BLK_TL-PIO.D_OUT[0]-&gt;PAD-D_OUT[0] open</port>
			</inputs>
			<outputs />
			<clocks />
			<block name="out:vec0[0]" instance="output[0]">
				<attributes />
				<parameters />
				<inputs>
					<port name="outpad">PAD.D_OUT[0]-&gt;D_OUT</port>
				</inputs>
				<outputs />
				<clocks />
			</block>
		</block>
	</block>
	<block name="out:vec1[4]" instance="BLK_TL-PIO[1]" mode="PAD_IS_OUTPUT">
		<inputs>
			<port name="D_OUT">vec1[4] open</port>
			<port name="OUT_ENB">open</port>
			<port name="CEN">open</port>
			<port name="LATCH">open</port>
		</inputs>
		<outputs>
			<port name="D_IN">open open</port>
			<port name="PACKAGE_PIN">open</port>
		</outputs>
		<clocks>
			<port name="INCLK">open</port>
			<port name="OUTCLK">open</port>
		</clocks>
		<block name="out:vec1[4]" instance="PAD[0]" mode="default">
			<inputs>
				<port name="D_OUT">BLK_TL-PIO.D_OUT[0]-&gt;PAD-D_OUT[0] open</port>
			</inputs>
			<outputs />
			<clocks />
			<block name="out:vec1[4]" instance="output[0]">
				<attributes />
				<parameters />
				<inputs>
					<port name="outpad">PAD.D_OUT[0]-&gt;D_OUT</port>
				</inputs>
				<outputs />
				<clocks />
			</block>
[... 2234 lines cut ...]
		</outputs>
		<clocks>
			<port name="INCLK">open</port>
			<port name="OUTCLK">open</port>
		</clocks>
		<block name="clk" instance="PAD[0]" mode="default">
			<inputs />
			<outputs>
				<port name="D_IN">input[0].inpad[0]-&gt;PAD-D_IN[0] open</port>
				<port name="PIN">open</port>
			</outputs>
			<clocks />
			<block name="clk" instance="input[0]">
				<attributes />
				<parameters />
				<inputs />
				<outputs>
					<port name="inpad">clk</port>
				</outputs>
				<clocks />
			</block>
		</block>
	</block>
</block>
```