f4pga/f4pga-arch-defs #216: vpr --pack -place --route results in invalid HLC

mcmasterg, 2018-07-31:
While most SymbiFlow testing is done with three separate VPR runs for pack, place, and route, fpga-tool-perf is using these separately. At some point these stopped working, instead requiring "vpr --pack --place" followed by "vpr --route".

Symptom:
```
$ icebox_hlc2asc top.hlc > my.asc
$ icetime -tmd lp8k my.asc
...
No path found!
```

Looking at top.hlc it seems that routing is missing. See for example this comment:
```
#Edge: (-3,-3) - ( 53518) SOURCE < 2> ( 53519)   OPIN          No HLC name (  1,  1) -- (  1,  1) No HLC name
```

Which should have route names there. Here's the issue in the actual directives:

```
    lutff_7 {
        # $abc$1182$n55

        out = 16'b1000000000000000

        # disable_dff
    }
```

When it should give something like:

```
    lutff_7 {
        # $abc$1182$n55

        # 53437 => 118343    OPIN => CHANY
        lutff_7/out -> local_g2_7
        # 53437 => 187196    OPIN => CHANY
        lutff_7/out -> span4_x18_g28_11
        # 53437 => 118351    OPIN => CHANY
        lutff_7/out -> local_g3_7
        # 118323 => 53413   CHANY => IPIN
        local_g0_3 -> lutff_7/in_2
        # 118324 => 53411   CHANY => IPIN
        local_g0_4 -> lutff_7/in_1
        # 118325 => 53409   CHANY => IPIN
        local_g0_5 -> lutff_7/in_0
        # 118326 => 53415   CHANY => IPIN
        local_g0_6 -> lutff_7/in_3
        out = 16'b1000000000000000

        # disable_dff
    }
```

elmsfu, 2019-03-22:
HLC no longer used. But this may still be a valid issue with running all the steps at once instead of separately. Will need to retest to see if this issue can be reproduced.

litghost, 2019-03-22:
I've actually independently seen issues around this.  I've in fact fixed at least one bug related to this caused by a double metadata load.  Could that explain what this issue is?

closed 2020-10-29 by litghost