f4pga/f4pga-arch-defs #262: XML dependencies are calculated before ntemplate subsitution

mithro, 2018-10-16:
The following `ntemplate.Nlut.pb_type.xml` which has an `xi:include` fails;
```xml
<pb_type name="{N}LUT" num_pb="1" xmlns:xi="http://www.w3.org/2001/XInclude">
 <!-- Normal LUT input/outputs -->
 <input  name="A1"   num_pins="1" />
 <input  name="A2"   num_pins="1" />
 <input  name="A3"   num_pins="1" />
 <input  name="A4"   num_pins="1" />
 <input  name="A5"   num_pins="1" />
 <input  name="A6"   num_pins="1" />
 <output name="O5"   num_pins="1" />
 <output name="O6"   num_pins="1" />

 <!-- Extended memory LUT pins -->
 <input  name="WA1"  num_pins="1" />
 <input  name="WA2"  num_pins="1" />
 <input  name="WA3"  num_pins="1" />
 <input  name="WA4"  num_pins="1" />
 <input  name="WA5"  num_pins="1" />
 <input  name="WA6"  num_pins="1" />
 <input  name="WA7"  num_pins="1" />
 <input  name="WA8"  num_pins="1" />
 <input  name="WE"   num_pins="1" />
 <input  name="DI1"  num_pins="1" />
 <input  name="DI2"  num_pins="1" />
 <output name="MC31" num_pins="1" />

 <xi:include href="../../slicel/{N}lut/{N}lut.pb_type.xml" xpointer="xpointer(pb_type/mode)" />
 <mode name="DISTRIBUTED_RAM">
  <pb_type name="RAM64X1D" num_pb="1" blif_model=".subckt RAM64X1D" class="memory">
   <clock  name="CLK1"          num_pins="1"  port_class="clock" />

   <!-- A1 - Read port -->
   <input  name="A1ADDR"        num_pins="6" port_class="address1" />
   <input  name="A1DATA"        num_pins="1" port_class="data_in1" />
   <!-- <output name="A1DATA"   num_pins="1"  port_class="data_out1" /> -->
   <!-- <input  name="A1WE"     num_pins="1"  port_class="write_en1" /> -->

   <!-- B1 - Write Port -->
   <input  name="B1ADDR"        num_pins="6" port_class="address2" />
   <input  name="B1DATA"        num_pins="1" port_class="data_in2" />
   <input  name="B1WEN"         num_pins="1" port_class="write_en2" />
   <!-- <output name="B1DATA"        num_pins="1" port_class="data_out2" /> -->
  </pb_type>
  <interconnect>
   <direct name="{N}LUT.A1" input="{N}LUT.A1" output="RAM64X1D.B1ADDR[0]" />
   <direct name="{N}LUT.A2" input="{N}LUT.A2" output="RAM64X1D.B1ADDR[1]" />
   <direct name="{N}LUT.A3" input="{N}LUT.A4" output="RAM64X1D.B1ADDR[2]" />
   <direct name="{N}LUT.A4" input="{N}LUT.A5" output="RAM64X1D.B1ADDR[3]" />
   <direct name="{N}LUT.A5" input="{N}LUT.A6" output="RAM64X1D.B1ADDR[4]" />
   <direct name="{N}LUT.A6" input="{N}LUT.A7" output="RAM64X1D.B1ADDR[5]" />
  </interconnect>
 </mode>
```

It causes the following cmake output;
```
tansell@tansell:~/github/SymbiFlow/symbiflow-arch-defs-a7/build$ cmake ..
CMake Error at make/file_targets.cmake:83 (get_target_property):
  get_target_property() called with non-existent target
  "file__home_tansell_github_SymbiFlow_symbiflow-arch-defs-a7_artix7_primitives_slicel_{N}lut_{N}lut.pb_type.xml".
Call Stack (most recent call first):
  make/file_targets.cmake:243 (append_file_dependency)
  artix7/primitives/slicem/Nlut/CMakeLists.txt:2 (add_file_target)

CMake Error at make/file_targets.cmake:83 (get_target_property):
  get_target_property() called with non-existent target
  "file__home_tansell_github_SymbiFlow_symbiflow-arch-defs-a7_artix7_primitives_slicel_{N}lut_{N}lut.pb_type.xml".
Call Stack (most recent call first):
  make/file_targets.cmake:96 (append_file_dependency)
  make/gen.cmake:321 (append_file_dependency)
  artix7/primitives/slicem/Nlut/CMakeLists.txt:9 (n_template)

CMake Error at make/file_targets.cmake:83 (get_target_property):
  get_target_property() called with non-existent target
  "file__home_tansell_github_SymbiFlow_symbiflow-arch-defs-a7_artix7_primitives_slicel_{N}lut_{N}lut.pb_type.xml".
Call Stack (most recent call first):
  make/file_targets.cmake:96 (append_file_dependency)
  make/gen.cmake:321 (append_file_dependency)
  artix7/primitives/slicem/Nlut/CMakeLists.txt:9 (n_template)

CMake Error at make/file_targets.cmake:83 (get_target_property):
  get_target_property() called with non-existent target
  "file__home_tansell_github_SymbiFlow_symbiflow-arch-defs-a7_artix7_primitives_slicel_{N}lut_{N}lut.pb_type.xml".
Call Stack (most recent call first):
  make/file_targets.cmake:96 (append_file_dependency)
  make/gen.cmake:321 (append_file_dependency)
  artix7/primitives/slicem/Nlut/CMakeLists.txt:9 (n_template)

CMake Error at make/file_targets.cmake:83 (get_target_property):
  get_target_property() called with non-existent target
  "file__home_tansell_github_SymbiFlow_symbiflow-arch-defs-a7_artix7_primitives_slicel_{N}lut_{N}lut.pb_type.xml".
Call Stack (most recent call first):
  make/file_targets.cmake:96 (append_file_dependency)
  make/gen.cmake:321 (append_file_dependency)
  artix7/primitives/slicem/Nlut/CMakeLists.txt:9 (n_template)

-- Configuring incomplete, errors occurred!
See also "/home/tansell/github/SymbiFlow/symbiflow-arch-defs-a7/build/CMakeFiles/CMakeOutput.log".
```

mithro, 2019-04-08:
@litghost - Is this still a problem?

kgugala, 2019-04-16:
@mithro yes, it seems it is - I got hit by that when working on generating all the xmls from verilog

PR #594 (2019-04-17): gen.cmake: handle generated verilog and xml deps
V2X and N_TEMPLATE now can handle dependency on a generated XML or Verilog file. The list of the dependencies can be passed via GENERATEDXMLDEPS and GENERATEDVERILOGDEPS for both n_template and v2x functions.

This may fix #262
closed 2019-04-17 by litghost