f4pga/f4pga-arch-defs #286: Fix timing based packing

litghost, 2018-12-19:
Some blocks in the current packing graph result in invalid timing paths that cause the clusterer to explode.  This is likely as simple as fixing broken timing paths and removing constant source from routing graph (see #285).

litghost, 2019-01-30:
Timing based packing is currently **disabled** via the options " --place_algorithm bounding_box --enable_timing_computations off".  Removing these options from https://github.com/SymbiFlow/symbiflow-arch-defs/blob/master/xc7/artix7.cmake#L69 will turn timing based packing back on.

When timing based packing is on, VPR crashes because outputs from the TIEOFF block have no clock.  This is actually correct, because these outputs are constants.  However the current use of TIEOFF is incorrect, and should be refactored to solve #285.

kgugala, 2019-01-31:
I re-enabled timing packing and vpr does not crash. Enabling that did not fix #284

litghost, 2019-01-31:
Double check that the arguments are present with make VERBOSE=1, sometimes the CMake variable cache gets in the way.

kgugala, 2019-01-31:
I double checked this. Timing based placing is enabled.

closed 2020-10-29 by litghost