f4pga/f4pga-arch-defs #290: Add Travis-CI support for 7-series

litghost, 2018-12-19:
Currently the RAM usage for the 7-series workflow is too large for Travis-CI.  Refactor the 7-series routing import logic to avoid using more than 4 GB of RAM.

litghost, 2019-04-05:
I believe the routing import can now be performed on travis, we should add rr graph generation for 7-series back to travis, combined with some parallelism to bring the runtime in check.

litghost, 2019-04-05:
Turns out the bigger problem is the 50 minute time limit on travis, and we are already close to that with current runs taking ~38 minutes.

mithro, 2019-04-05:
You can have multiple runs at the same time - see https://github.com/SymbiFlow/prjxray/blob/master/.travis.yml#L21-L34 for example.

If we can split this into ice40, testarch and xc7 would it be fast enough?

litghost, 2019-04-05:
> You can have multiple runs at the same time - see https://github.com/SymbiFlow/prjxray/blob/master/.travis.yml#L21-L34 for example.
> 
> If we can split this into ice40, testarch and xc7 would it be fast enough?
> 
> > [image][image] GitHub**[SymbiFlow/prjxray](https://github.com/SymbiFlow/prjxray)**Documenting the Xilinx 7-series bit-stream format. - SymbiFlow/prjxray

Potentially, and problably valuable anyways to cut down on CI time.  Probably worth attacking splitting the arches first, then add xc7 as a follow up.  I've thought about having explicit arch support flags, and this would be a good outgrowth of that idea.

litghost, 2019-04-10:
With kokoro CI running, this is obsolete.

closed 2019-04-10 by litghost