f4pga/f4pga-arch-defs #293: Improve timing model support for 7-series

litghost, 2018-12-19:
Currently the timing model is total nonsense because of bad constants for segments and pips.

mithro, 2018-12-19:
There is a timing fuzzer in Project X-Ray which generates some timing data. It can be found at https://github.com/SymbiFlow/prjxray/tree/master/fuzzers/007-timing

kgugala, 2019-03-07:
BEL timings seem to easily obtainable using `report_property` and `get_timing_arcs` functions.

e.g for LUT6 I have:

```
report_property -all [lindex [get_timing_arcs -of_objects [get_cells  lut]] 0]
Property             Type    Read-only  Value
CLASS                string  true       timing_arc
DELAY_FAST_MAX_FALL  double  true       0.056
DELAY_FAST_MAX_RISE  double  true       0.056
DELAY_FAST_MIN_FALL  double  true       0.045
DELAY_FAST_MIN_RISE  double  true       0.045
DELAY_MAX_FALL       double  true       0.124
DELAY_MAX_RISE       double  true       0.124
DELAY_MIN_FALL       double  true       0.045
DELAY_MIN_RISE       double  true       0.045
DELAY_SLOW_MAX_FALL  double  true       0.124
DELAY_SLOW_MAX_RISE  double  true       0.124
DELAY_SLOW_MIN_FALL  double  true       0.100
DELAY_SLOW_MIN_RISE  double  true       0.100
FROM_PIN             pin     true       lut/I0
IS_DISABLED          bool    true       0
IS_USER_DISABLED     bool    true       0
LIB_CELL             string  true       xt_artix7/LUT6
NAME                 string  true       {lut/I0 --> lut/O [combinational] }
SENSE                string  true       negative_unate
TO_PIN               pin     true       lut/O
TYPE                 enum    true       combinational
```

while for FDRE:
```
report_property -all [lindex [get_timing_arcs -of_objects [get_cells FDRE_inst]] 2]
Property             Type    Read-only  Value
CLASS                string  true       timing_arc
DELAY_FAST_MAX_FALL  double  true       0.175
DELAY_FAST_MAX_RISE  double  true       0.175
DELAY_FAST_MIN_FALL  double  true       0.141
DELAY_FAST_MIN_RISE  double  true       0.141
DELAY_MAX_FALL       double  true       0.456
DELAY_MAX_RISE       double  true       0.456
DELAY_MIN_FALL       double  true       0.141
DELAY_MIN_RISE       double  true       0.141
DELAY_SLOW_MAX_FALL  double  true       0.456
DELAY_SLOW_MAX_RISE  double  true       0.456
DELAY_SLOW_MIN_FALL  double  true       0.367
DELAY_SLOW_MIN_RISE  double  true       0.367
FROM_PIN             pin     true       FDRE_inst/C
IS_DISABLED          bool    true       0
IS_USER_DISABLED     bool    true       0
LIB_CELL             string  true       xt_artix7/FDRE
NAME                 string  true       {FDRE_inst/C --> FDRE_inst/Q [Reg Clk to Q] }
SENSE                string  true       non_unate
TO_PIN               pin     true       FDRE_inst/Q
TYPE                 enum    true       Reg Clk to Q
```

I'll create a fuzzer for getting all of those

mithro, 2019-03-07:
Please take a look at https://github.com/SymbiFlow/symbiflow-arch-defs/pull/165/files for how I was going about it for the ice40.

Ideally I think we want to end up with a SDF file which describes the timing internally of the SLICEL.

kgugala, 2019-03-07:
Do we have any tool that can update arch.xml with data from SDF?

mithro, 2019-03-07:
Not yet.

mithro, 2019-03-07:
@kgugala - VtR however will produce a SDF file for a place and routed design at the end and a number of other tools accept SDF as input.

elmsfu, 2019-04-12:
#55 relates to this. I think specify blocks are parsed by yosys but ignored (I have not confirmed this)

#574 is the start importing before `v2x`

Here is a [working document](https://docs.google.com/document/d/1nucKg8i07MI8f6lbZhJKlSSrC33_exBNeC3kt2sgDYY) to capture some thoughts.

elmsfu, 2019-04-12:
@kgugala Here is an iCE40 SDF with a [clear violation](https://gist.github.com/elmsfu/0a3c87247496c57b488a2a2ce789a22c#file-timings_hx1k-sdf-L213) of the SDF standard around the PLL.

closed 2019-10-22 by litghost