f4pga/f4pga-arch-defs #294: Move VPR specific yosys tech maps into symbiflow-arch-defs

litghost, 2018-12-19:
https://github.com/SymbiFlow/yosys/pull/5 put a lot of VPR specific stuff into the yosys techlib. https://github.com/SymbiFlow/yosys/pull/5#issuecomment-442715959 is right that these changes would break non-VPR flows, and the techmaps don't belong in yosys anyways.  Move the VPR only techmaps into symbiflow-arch-defs and revert the techmaps in yosys's xilinx flow.

mithro, 2018-12-19:
The idea is to do a pass **after** the normal yosys xilinx flow which is vtr specific.

litghost, 2019-02-12:
https://github.com/SymbiFlow/symbiflow-arch-defs/blob/master/xc7/techmap/cells_map.v

mkurc-ant, 2019-02-13:
Could you please assign me to this issue? I'am currently working on it and getting closer to a solution of the problem.

elmsfu, 2019-02-13:
@mithro looks like we need to add @mkurc-ant to an appropriate team to assign this. There may be other recent people we should add to a team.

mithro, 2019-02-14:
@mkurc-ant - Done! This issue is now assigned to you.

mkurc-ant, 2019-02-15:
Great, thanks!

mkurc-ant, 2019-02-19:
I've been struggling with this issue for some time and I do not have a complete solution. The thing is that there is no possibility to map Vivado primitives to VPR primitives just by using a techmap. That is because:

1. Carry chains
Vivado allows implicit carry output connections. The decision whether to go to the fabric or to another carry chain is make after synthesis in the P&R pass. However the VPR requires explicit connections. Infering those would require adding some C code to the Yosys

2. LUTs
Making LUT7 and LUT8 requires adding explicitly MUX7F and/or MUX8F which connects LUT6s to form a bigger one. This is ok with Vivado. From what I understand VPR uses $lut primitives from Yosys and figures out how and where to put them (right?). So when I have a $lut with WIDTH=7 and I map it to 2xLUT6+MUX7F for Vivado I cannot map this back to $lut with WIDTH=7 for the VPR using just techmap.

3. Potential future problems with #373 (Slice wide FF inverters)
Yosys recognizes FFs with inverters (internally), but when I map them to those recognized by Vivado (which do not allow explicit negation of their inputs) I loose information about the inversion. The inversion becomes part of logic connected to FFs so it cannot be mapped back for VPR.

It seems to me that without adding additional functionality to the VPR we cannot make it completely independent from the Yosys. There will always have to be a different flow for Vivado and VPR in Yosys as it is now.

litghost, 2019-02-19:
>     2\. Making LUT7 and LUT8 requires adding explicitly MUX7F and/or MUX8F which connects LUT6s to form a bigger one. This is ok with Vivado. From what I understand VPR uses $lut primitives from Yosys and figures out how and where to put them (right?). So when I have a $lut with WIDTH=7 and I map it to 2xLUT6+MUX7F for Vivado I cannot map this back to $lut with WIDTH=7 for the VPR using just techmap.

LUTs should be able to use the same techmap, VPR also requires explicit instancing of any LUT greater than 6 inputs.

litghost, 2019-02-19:
> 3\. Potential future problems with #373 (Slice wide FF inverters)
>     Yosys recognizes FFs with inverters (internally), but when I map them to those recognized by Vivado (which do not allow explicit [...]

I double checked #373, only clock inversion is actually configuration.  How is the currently Yosys to Vivaod clock inversion handled?

litghost, 2019-02-19:
In the case of 1 and 3, it seems there is a straight forward place in the synthesis flow where the divergence occurs.  The original intention of this issue was to move as much of the techmaping out of the yosys repo into the symbiflow-arch-defs.  That simply means stopping the yosys flow that is part of synth_xilinx at a place prior to the divergence.  So in the case of FF's, the clock inversion parameter would still be present in a VPR oriented synth_xilinx flow.  Same with carry chain.

mkurc-ant, 2019-02-20:
OK, So I'll try to move as much out of Yosys as possible. Keep in mind that I'll have to modify the flow of the @synth_xilinx@ macro. Unfortunatelly there is no a single point of divergence.

An alternative is to create a separate TCL script (or modify xc7/yosys/synth.tcl) for VPR and move the whole synthesis flow there. Therefore we will not have to use synth_xilinx at all.

mkurc-ant, 2019-02-20:
I've made two pull requests. One for SymbiFlow:
https://github.com/SymbiFlow/symbiflow-arch-defs/pull/390

And the second one one for Yosys
https://github.com/SymbiFlow/yosys/pull/8

mithro, 2019-04-08:
@litghost / @mkurc-ant - Has this been solved now? I think so?

litghost, 2019-04-08:
Ya

closed 2019-04-08 by mithro