f4pga/f4pga-arch-defs #311: basys3 counter example does not build

kgugala, 2019-01-03:
It fails with:

```
Error 1: 
Type: Blif file
File: symbiflow-arch-defs/build/artix7/tests/counter/counter/artix7-xc7a50t-roi-virt-xc7a50t-test/top.eblif
Line: 12
Message: Failed to find matching architecture model for 'CYINIT_CONSTANTS'
```

kgugala, 2019-01-03:
I added the `CYINIT_CONSTANT` models from https://github.com/litghost/symbiflow-arch-defs/tree/artix7_startup/artix7/primitives/slicel

now the build fails with `Failed to find matching architecture model for 'FDRE'`

kgugala, 2019-01-03:
the `FDRE` issue is caused by too old yosys version downloaded by conda. Building with current SymbiFlow/yosys master does not generate this error

litghost, 2019-01-03:
Ya, those errors are caused by an old yosys techmap.  #294 was created to move some of the tech mapping definitions from yosys to symbiflow to mitigate this issue somewhat.

litghost, 2019-01-03:
I think tim kicked off a conda package rebuild with https://github.com/SymbiFlow/conda-packages/commit/5f02998f8cbf24bffa5564fe7f34f07ccf0749ea so hopefully this fixes itself.

litghost, 2019-01-03:
FYI, until #294 is resolved, I recommend setting the env var YOSYS to point to a local yosys executable.

mithro, 2019-01-04:
I believe the Yosys in conda should be up to date now. Can someone please
check if it works?

kgugala, 2019-01-04:
Yosys from conda is now correct.

PR #314 (2019-01-04): Basys3 counter demo fix
This PR partially fixes #311. The build passes up to `top.frames` generation and it fails on patching the bistream with:

```
[100%] Generating counter/artix7-xc7a50t-roi-virt-xc7a50t-test/top.bit
frame address 0x2049e because it was not found in frames.
```

The changes in this PR mostly fix the `pb_types` definitions so they match latest prjxray-db
closed 2019-01-04 by litghost