f4pga/f4pga-arch-defs #312: make env does not build the environment

kgugala, 2019-01-03:
invoking `make env` did the following:

```
make env 
git submodule init
git submodule update --init --recursive
mkdir -p build && cd build && cmake ..
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.5.4", minimum required is "3") 
-- Found PythonInterp: /usr/bin/python3 (found version "3.5.4") 
CMake Deprecation Warning at third_party/prjxray/third_party/yaml-cpp/CMakeLists.txt:9 (cmake_policy):
  The OLD behavior for policy CMP0012 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at third_party/prjxray/third_party/yaml-cpp/CMakeLists.txt:13 (cmake_policy):
  The OLD behavior for policy CMP0015 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- Configuring done
-- Generating done
-- Build files have been written to: .../symbiflow-arch-defs/build
```

the env is actually build when an example is build

litghost, 2019-01-03:
This was the intended behavior of the root level "make env".  If we want the conda steps to be run, the "env" CMake target will complete the setup.

kgugala, 2019-01-05:
right. Thanks for clarification.

closed 2019-01-05 by kgugala