f4pga/f4pga-arch-defs #338: Merge the channel.py and channel2.py files in utils

mithro, 2019-01-24:
We have both;
 * https://github.com/SymbiFlow/symbiflow-arch-defs/blob/master/utils/lib/rr_graph/channel.py
 * https://github.com/SymbiFlow/symbiflow-arch-defs/blob/master/utils/lib/rr_graph/channel2.py

It would be good to merge them and add a decent test suite.

Fleker, 2019-02-18:
Good evening,
I'm interested in getting involved in this project, and I can work on this issue.

Fleker, 2019-02-18:
Looks like channel2.py is only used by graphs2.py

mithro, 2019-02-19:
Merging grpahs.py and graphs2.py is also on the TODO list.

mithro, 2019-02-19:
There might also be useful information [in this Google doc](https://docs.google.com/document/d/17QBzXAul8ioGL7WeTXk0awwspGIDZBZgTYUKy5oT_Gg/edit?usp=drivesdk)

Fleker, 2019-03-18:
The easiest way to do this would be to just copy and paste, although it seems like Channel2.py was created due to some deep-seated bugs in Channel.py track layout. Would it make more sense to fix the Channel.py bug, or just merge the files?

litghost, 2019-03-18:
channel.py is simply wrong, so I wouldn't both to fix it.

mithro, 2019-03-18:
@litghost Do you remember what made you think that channel.py was wrong?

litghost, 2019-03-18:
> @litghost Do you remember what made you think that channel.py was wrong?

From channels2.py:

```
channel.py is actually broken for some track configuration, and generates
an excessive number of dummy tracks to fill empty space (channel.Channel = >2M
versus channel2.Channel ~70k).
```

There is also the fact that channels.py is ~1141 lines long and channels2.py is 40 lines long.

mithro, 2019-03-18:
channels.py is missing a lot of the pretty printing functionality,
https://github.com/SymbiFlow/symbiflow-arch-defs/blob/7ad5525b8b5d8128ef5aed84bf000689712821f6/utils/lib/rr_graph/channel.py#L1051-L1068

https://github.com/SymbiFlow/symbiflow-arch-defs/blob/7ad5525b8b5d8128ef5aed84bf000689712821f6/utils/lib/rr_graph/channel.py#L1113-L1122

litghost, 2019-03-18:
I don't there is value in pretty printing (to text) channels when you are talking about hundreds and thousands of channels.  If we need to visualize channels, we already have VTR visualization.  If want a better visualization, we should probably enhance VTR's GUI or create something of our own that has zoom capabilities.

Fleker, 2019-03-19:
@litghost there is not a `channel.Channel` class. Of three main methods in channel2 (`place_track`, `pack_tracks`, and `fill_empty`), only `fill_empty` is implemented in `channel.ChannelGrid`, and used in the `channel.Channels` class.