f4pga/f4pga-arch-defs #339: Merge the graph.py and graph2.py files in utils

mithro, 2019-01-24:
We have both;
 * https://github.com/SymbiFlow/symbiflow-arch-defs/blob/master/utils/lib/rr_graph/graph.py
 * https://github.com/SymbiFlow/symbiflow-arch-defs/blob/master/utils/lib/rr_graph/graph2.py

It would be good to merge them and add a decent test suite.

Fleker, 2019-03-18:
It looks like the files do a lot of similar things, but differently. For example, graph2.py defines a few enums like SwitchType as numbers, whereas SwitchType in graph.py uses strings. Is there a preference for this?

litghost, 2019-03-18:
In all cases prefer graph2 over graph.

elmsfu, 2019-03-18:
To give some background: graph is used for the ice40 portion and is tightly coupled with the XML. graph2 is for xc7 and avoids being tied with the XML format and is more optimised for larger FPGAs. I think the the optimizations (and separation from XML) are the reasons @litghost says prefer graph2.

It may make sense to have some unit tests at the higher level to allow quicker iteration. I'm actually adding some to for [ice40_import_layout_from_icebox.py](https://github.com/SymbiFlow/symbiflow-arch-defs/blob/013388bf79c452659ded69283d5dc65e3f2d713d/ice40/utils/ice40_import_layout_from_icebox.py). So maybe some tests around [ice40_import_routing_from_icebox.py](https://github.com/SymbiFlow/symbiflow-arch-defs/blob/013388bf79c452659ded69283d5dc65e3f2d713d/ice40/utils/ice40_import_routing_from_icebox.py#L90) is a good place to start.

I think this is a bigger task and could easily be broken down.