f4pga/f4pga-arch-defs #372: VTR cannot pack slicem.carrychain

kgugala, 2019-02-11:
This can be reproduced by uncommenting the fifth counter in https://github.com/SymbiFlow/symbiflow-arch-defs/blob/master/xc7/tests/chain_packing/counter.v#L7
and using VPR from this PR https://github.com/SymbiFlow/vtr-verilog-to-routing/pull/9

This will pack the fifth counter into SLICEM.CARRYCHAIN and VPR packer fails with:
```
Error 1: 
Type: Packing
File: vtr-verilog-to-routing/vpr/src/pack/cluster.cpp
Line: 2029
Message: Can not find any logic block that can implement molecule.
	Pattern BLK_TI-CLBLM_R.BLK_IG-SLICEM.CARRYCHAIN $auto$alumacc.cc:474:replace_alu$29.slice[0].carry
```
This happens only if a chain is packed into SLICEM

acomodi, 2019-02-11:
I have hacked this error by removing the CLBLM_[LR] definition from the `device_type` instantiation in `cmake` as in the example below:

```
add_xc7_device_define_type(
  ARCH artix7
  DEVICE xc7a50t
  ROI_DIR ${PRJXRAY_DB_DIR}/artix7/harness/basys3/swbut
  TILE_TYPES INT_R INT_L CLBLL_R CLBLL_L BRAM_L
)
```

This has the effect of not including `DRAM` blocks which are connected to the CLBLM_[LR] tiles.

litghost, 2019-02-11:
Ya, you completely removed CLBLM_[LR] tiles from the arch, which is not a solution.

acomodi, 2019-02-11:
Yes indeed, It Is just a temporary workaround, it should be properly solved

acomodi, 2019-02-19:
I have been studying this problem. Follows the steps to reproduce it:

- VPR version: https://github.com/SymbiFlow/vtr-verilog-to-routing/pull/9
- Symbiflow-arch-defs: 79ab13f2b2ace5009b14fb11134f4e2a72fa9d63
- VPR options in [arch_define.cmake](https://github.com/SymbiFlow/symbiflow-arch-defs/blob/4b6651608ab699a1a56bfc6a491c327e97ffd5db/xc7/make/arch_define.cmake#L27) to be added: `--clustering_pin_feasibility_filter off`
- in `symbiflow-arch-defs/build` run `make chain_packing_basys3_route_echo`. (prior to this the [counter.v](https://github.com/SymbiFlow/symbiflow-arch-defs/blob/master/xc7/tests/chain_packing/counter.v) and [basys3.pcf](https://github.com/SymbiFlow/symbiflow-arch-defs/blob/master/xc7/tests/chain_packing/basys3.pcf) files have to be modified to allow 5 counters by commenting out the relative lines)

By digging in VPR I have noticed that the issue is in the `modes` of the `SLICEM`. The problem should be that a `CLBLM_[RL]` that has to be packed contains the `SLICEM` that can have different `modes` such  as `DRAMs`, `LUTs` or `DRAM128`. 
What happens is that there is a change in the `modes` within `PB blocks` of a cluster that has to be routed, making the actual routing unfeasible. The changes apported with https://github.com/SymbiFlow/vtr-verilog-to-routing/pull/6 should deal with this problem, in fact, an error is risen when checking the `modes` of `SLICEM` blocks.
The interesting file that has to be checked and which is responsible for this issue is [cluster_router.cpp](https://github.com/SymbiFlow/vtr-verilog-to-routing/blob/master%2Bwip/vpr/src/pack/cluster_router.cpp).
I need to further check and spot the place where the modes are assigned to the `PBs` in a cluster. Probably solving the `modes` assignment will solve this issue, allowing us to use `SLICEMs`, both in `DRAM` and `LUT` mode.

mithro, 2019-02-20:
@acomodi 
This sounds related to https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/365

mithro, 2019-02-20:
IIRC The `clustering_pin_feasibility_filter` is just a performance enhancement. It will stop the clusterer from doing more expensive routing checks if there are not enough top level pins to satisfy adding a molecule to the cluster (which mean the routing checks will ultimately fail).

mithro, 2019-02-20:
@acomodi Oh btw, the idea with the [testarch](https://github.com/SymbiFlow/symbiflow-arch-defs/tree/master/testarch) directory is to enable very simple example architectures which have features like this which we can make sure VPR continues to support properly.

acomodi, 2019-03-20:
Closing. Fixed through https://github.com/SymbiFlow/vtr-verilog-to-routing/pull/29

closed 2019-03-20 by acomodi