f4pga/f4pga-arch-defs #377: Extract parameters from PDF specifications

mjasperse, 2019-02-13:
Following from idea #149, also extract parameter specifications from the PDF file

mjasperse, 2019-02-13:
I've modified the script from PR #374 to scrape the "Available Attributes" table for each relevant module, which works but has new edge cases to work around.

How do you want the parsed data represented in the output? Maybe something of the following form?
`parameter $NAME = $DEFAULT;  // $TYPE, allowed values: $ALLOWED`

elmsfu, 2019-02-13:
I think that is a reasonable approach. It captures all the information. In the future we may need to change it once we use it for validation.

mithro, 2019-02-14:
In some of the other Verilog we have used this form;

```verilog

(* MODE="REGISTERED_NONE; REGISTERED_IN; REGISTERED_OUT; REGISTERED_INOUT; REGISTERED_PARTIAL" *)
module blah (clk, a, b, m, out);

	parameter MODE = "REGISTERED_INOUT";

 	/* Register modes */
	generate
		if (MODE == "REGISTERED_NONE") begin
			dsp_combinational dsp_int_comb (.clk(clk), .a(a), .b(b), .m(m), .out(out));
		end if (MODE == "REGISTERED_IN") begin
			dsp_in_registered dsp_int_regi (.clk(clk), .a(a), .b(b), .m(m), .out(out));
		end if (MODE == "REGISTERED_OUT") begin
			dsp_out_registered dsp_int_rego (.clk(clk), .a(a), .b(b), .m(m), .out(out));
		end if (MODE == "REGISTERED_INOUT") begin
			dsp_inout_registered dsp_int_regio (.clk(clk), .a(a), .b(b), .m(m), .out(out));
		end if (MODE == "REGISTERED_PARTIAL") begin
			dsp_partial_registered dsp_int_part (.clk(clk), .a(a), .b(b), .m(m), .out(out));
		end
	endgenerate

```

mithro, 2019-02-14:
We might also want to consider sphinx style strings.

The two most popular formats are;
 * NumPy style - https://numpydoc.readthedocs.io/en/latest/format.html#docstring-standard
 * Google Style - https://google.github.io/styleguide/pyguide.html#383-functions-and-methods

Then there is also Doxygen style comments too...

mithro, 2019-02-14:
We should probably look at generating a `generate` statement which makes sure the parameters are set to valid values?
```verilog
generate
if (CONDITION > MAX_ALLOWED /* your condition check */ ) begin
    $error("%m ** Illegal Condition ** CONDITION(%d) > MAX_ALLOWED(%d)", CONDITION, MAX_ALLOWED);
end
endgenerate
```

mithro, 2019-02-14:
I wonder if there should be two steps. One which extracts the data from the pdf files into something like csv and then a second step which takes that input and generates Verilog / VHDL / other language outputs?

mjasperse, 2019-02-14:
An intermediate representation might be sensible because there are two distinct tasks here - one being parsing the PDF document and the other being turning that into a range of complex outputs. I'm generally not a fan of CSV in general because of the lack of structure, and here some attributes accept a range of values, others have specific choices. I am using XML for testing/debugging the script (see generate_xml()) which is one option, but I can adapt it to whatever you prefer.

mithro, 2019-02-14:
No really strong preference around format. Any of XML, json, python pickle or CSV would be fine I think?

mjasperse, 2019-02-17:
This ended up being much more difficult than I anticipated compared to the previous parser, but I think it works now. To handle variations among the table entries the parser had to become much smarter, and then it became possible to extract the descriptions as well.

The original brief was just to get the minimum data to check compatibility, but the value checking proposed above requires more than that. So the intermediate XML contains all the data scraped from the PDF for verification, and the output verilog is just names and defaults. More involved verilog verification could be added by transforming the XML as required.

elmsfu, 2019-03-19:
This looks fixed in #374. Reopen if I'm incorrect.

Thanks again @mjasperse :star:

closed 2019-03-19 by elmsfu