f4pga/f4pga-arch-defs #382: The xc7/tests/counter example does not work from the current master branch (4b66516)

mkurc-ant, 2019-02-15:
The project builds but does not work on the actual hardware. All 4 LEDSs are off. Simulation of `top_synth.v` works fine. The last commit I've managed to successfully build and run is 79ab13f. I don't know if any other tests are affected. 

For all my tests I've set the `LOGIC_DELAY` parameter to 0

I've checked commits between 79ab13f and 4b66516 (master), here is the summary:

4b66516 (master):

The xc7/tests/counter builds but does not work. Got all 4 LEDs unlit.

50dca48:

Same as above.

ae9a635:

The build system fails with an error:

```
[ 66%] Generating pin_assignments.json
[ 66%] Built target file__home_mkurc_Repos_google-symbiflow-arch-defs_xc7_archs_artix7_channels.json
2019-02-15 17:02:28.175487 Reading channel data
Traceback (most recent call last):
  File "/home/mkurc/Repos/google-symbiflow-arch-defs/xc7/utils/prjxray_assign_tile_pin_direction.py", line 290, in <module>
    main()
  File "/home/mkurc/Repos/google-symbiflow-arch-defs/xc7/utils/prjxray_assign_tile_pin_direction.py", line 48, in main
    channels = json.load(f)
  File "/home/mkurc/Repos/google-symbiflow-arch-defs/build/env/conda/lib/python3.7/site-packages/simplejson/__init__.py", line 461, in load
    use_decimal=use_decimal, **kw)
  File "/home/mkurc/Repos/google-symbiflow-arch-defs/build/env/conda/lib/python3.7/site-packages/simplejson/__init__.py", line 518, in loads
    return _default_decoder.decode(s)
  File "/home/mkurc/Repos/google-symbiflow-arch-defs/build/env/conda/lib/python3.7/site-packages/simplejson/decoder.py", line 370, in decode
    obj, end = self.raw_decode(s)
  File "/home/mkurc/Repos/google-symbiflow-arch-defs/build/env/conda/lib/python3.7/site-packages/simplejson/decoder.py", line 400, in raw_decode
    return self.scan_once(s, idx=_w(s, idx).end())
simplejson.errors.JSONDecodeError: Invalid control character '\n' at: line 54604359 column 16 (char 945489370)
xc7/archs/artix7/CMakeFiles/file__home_mkurc_Repos_google-symbiflow-arch-defs_xc7_archs_artix7_pin_assignments.json.dir/build.make:243: recipe for target 'xc7/archs/artix7/pin_assignments.json' failed
make[3]: *** [xc7/archs/artix7/pin_assignments.json] Error 1
CMakeFiles/Makefile2:30763: recipe for target 'xc7/archs/artix7/CMakeFiles/file__home_mkurc_Repos_google-symbiflow-arch-defs_xc7_archs_artix7_pin_assignments.json.dir/all' failed
make[2]: *** [xc7/archs/artix7/CMakeFiles/file__home_mkurc_Repos_google-symbiflow-arch-defs_xc7_archs_artix7_pin_assignments.json.dir/all] Error 2
CMakeFiles/Makefile2:36320: recipe for target 'xc7/tests/counter/CMakeFiles/counter_basys3_bin.dir/rule' failed
make[1]: *** [xc7/tests/counter/CMakeFiles/counter_basys3_bin.dir/rule] Error 2
Makefile:382: recipe for target 'xc7/tests/counter/CMakeFiles/counter_basys3_bin.dir/rule' failed
make: *** [xc7/tests/counter/CMakeFiles/counter_basys3_bin.dir/rule] Error 2
```

79ab13f:

The xc7/tests/counter test works fine.

litghost, 2019-02-15:
You need a new ROI and database.  @mithro, status?

mithro, 2019-02-18:
I've pushed the new database but not the new harness yet.

Will push the harness tomorrow.

closed 2019-02-22 by litghost