f4pga/f4pga-arch-defs #416: Testing and verify support for using all 8 SLICE_[LM] carry chains

litghost, 2019-03-04:
Each of the 4 CLB tile types has 2 carry chains.  Currently the second chain in each tile is disabled.  Once https://github.com/SymbiFlow/vtr-verilog-to-routing/issues/17 and https://github.com/SymbiFlow/symbiflow-arch-defs/issues/415 are completed:

- [ ] Add test to verify that all 8 changes work as expect

acomodi, 2019-03-07:
SLICEM carrychain issue is still not fixed. To have it to correctly work https://github.com/SymbiFlow/symbiflow-arch-defs/issues/372 has to be solved first

acomodi, 2019-05-13:
@litghost can this be closed? If so we can also close https://github.com/SymbiFlow/symbiflow-arch-defs/issues/417

litghost, 2019-05-13:
I don't think we can close this until we've committed the equiv tiles and tested it.

acomodi, 2019-10-16:
@litghost I believe this can be closed?

closed 2020-10-29 by litghost