f4pga/f4pga-arch-defs #422: Add support for BUFGCTRL primitive

litghost, 2019-03-04:
Add a pb_type for BUFGCTRL.   This is required for global clock network placement.  BUFHCE should *not* start as placeable, plan is to route through BUFHCE.

acomodi, 2019-08-01:
I have started to deal with this issue.

I have a couple of questions:

- There are multiple modes in which a BUFGCTRL site can be instantiated. I have started to model the xml model/pb_type and I have currently modeled the `BUFGCTRL` and `BUFG` primitives. There is no need (for now) to support all the possible modes, right?
- Until the PLL is modeled, the input clock will be driven from an IO pin, correct? 
- There are two `CLK_BUFG_[(TOP)(BOT)` tiles, each one of them containing 16 `BUFGCTRL` sites. The problem is that each signal incoming/outgoing to/from the CLK_BUFG tile is relative to a specific site. Is it correct to have something like the following to model the input/outputs of `CLK_BUFG`?:

```xml
<pb_type name="CLK_BUFG_TOP">
  <input name="BUFGCTRL0_I1" .../>
  ...
  <input name="BUFGCTRL1_I1" .../>
  ...
  <input name="BUFGCTRL2_I1" .../>
  ...
  <input name="BUFGCTRL3_I1" .../>
  ...
  <input name="BUFGCTRL4_I1" .../>
  ...
  <pb_type name="CLK_BUFG" num_pb="16">
    <input name="I1">
    ...
  </pb_type>
  <interconnect>
    <direct name="I1[0]" input="BUFGCTRL0_I1" output"CLK_BUFG[0].I1">
    ...
  </interconnect>
</pb_type>
```

- How to make the `BUFHCE` not placeable but routable? The idea I had is to have pb_types representing `BUFHCEs` as children of `CLK_BUFG` tiles, but I am unsure whether this could be a good approach.

litghost, 2019-08-01:
> * There are multiple modes in which a BUFGCTRL site can be instantiated. I have started to model the xml model/pb_type and I have currently modeled the `BUFGCTRL` and `BUFG` primitives. There is no need (for now) to support all the possible modes, right?

Given that `BUFG` is a subset of `BUFGCTRL`, I recommend we only implement `BUFGCTRL`, and techmap `BUFG` to `BUFGCTRL`.

litghost, 2019-08-01:
> * Until the PLL is modeled, the input clock will be driven from an IO pin, correct?

The current ROI models the clock as the output of a BUFH.  In order to implement BUFG, you need to have an ROI (or ROI-less) setup.  I recommend as the first step to just create a BUFGCTRL leaf pb_type and yosys techmap for BUFG.

litghost, 2019-08-01:
> * There are two `CLK_BUFG_[(TOP)(BOT)` tiles, each one of them containing 16 `BUFGCTRL` sites. The problem is that each signal incoming/outgoing to/from the CLK_BUFG tile is relative to a specific site. Is it correct to have something like the following to model the input/outputs of `CLK_BUFG`?:

I think it is time to get equiv tiles on upstream and then use them here.  In particular, I believe BUFG's should be implemented as ``capacity`` on tile, e.g. the cluster is just the 1 BUFG, and the placer can place in any of the 32 BUFG placement locations (two stacks of 16).  The routing import will need to be taught how to handle this.

litghost, 2019-08-01:
> * How to make the `BUFHCE` not placeable but routable? The idea I had is to have pb_types representing `BUFHCEs` as children of `CLK_BUFG` tiles, but I am unsure whether this could be a good approach.

There are couple different ways to handle this.  The child approach is not suitable because of how `tall` 7-series parts work.  I would talk to the VPR devs, but I think the way to handle route through BUFH's is by creating a BUFH site, and then connecting the SOURCE to the SINK IPIN, and adding the correct FASM features.  Again, I would talk to the VPR devs about this.  They have some undocumented clock modelling stuff that may be helpful here.

acomodi, 2019-08-22:
@litghost I have started to look back at this issue.
I have a question: how does the https://github.com/YosysHQ/yosys/pull/1292 affects this issue?

litghost, 2019-08-22:
YosysHQ/yosys#1292 is a synthesis change.  For now, focus on the place/route implications, e.g. assume that the BUFG/BUGFCTRL already exists and just needs to be place/routed by VPR.

acomodi, 2019-08-23:
@litghost Ok. For now I am adding BUFGCTRL to the architecture. What I am planning to get is something looking like this (I still need to figure out how to treat this during the `routing_import` phase).

```xml
<pb_type name="BUFGCTRL_TOP" capacity="16">
  <input/outputs/clocks />
  <pb_type name="BUFGCTRL_VPR" num_pb="1">
    <input/outputs/clocks />
  </pb_type>
  <interconnect .../>
</pb_type>

<pb_type name="BUFGCTRL_BOT" capacity="16">
  <input/outputs/clocks />
  <pb_type name="BUFGCTRL_VPR" num_pb="1">
    <input/outputs/clocks />
  </pb_type>
  <interconnect .../>
</pb_type>
```

Now, I have seen that there is no `capacity` support right now during the `tile_import` and I am trying to add that. The problem is that, `pin_assignments` is related to the whole `CLK_BUFG_TOP/BOT_R` (which I have named `BUFGCTRL_BOT/TOP` in the example) containing all the 16 `BUFGCTRL` sites.
What I am trying to do is to declare the `BUFGCTRL` pb_type only once, but the fact is that the pins of BUFGCTRL do not match with the ones of the corresponding tile.

Instead, I have also tried to generate 16 `pb_type.xml` from a `BUFGCTRL` template, and the tile_import correctly produced an architecture with 16 pb_types for each `CLK_BUFG_R` tile.

I am not sure on what would be the best approach here. I guess that, to have a solution as suggested [here](https://github.com/SymbiFlow/symbiflow-arch-defs/issues/422#issuecomment-517350509), I need also to modify how the `pin_assignments` file is generated to take into account the pins issue.

acomodi, 2019-08-23:
I am thinking that maybe this could be solved by splitting the `CLK_BUFG_TOP/BOT` tiles as it happened for the CLBL[ML].

litghost, 2019-09-04:
> I am thinking that maybe this could be solved by splitting the `CLK_BUFG_[TOP/BOT]` tiles as it happened for the `CLBL[ML]`.

The site split is not the correct solution.  I think this is a good case for equivalent tiles support.  Is this possible:
1. `CLK_BUFG_TOP` and `CLK_BUFG_BOT` top level port names match their site pin wire names (e.g. `CLK_BUFG_BUFGCTRL0_O`, `CLK_BUFG_BUFGCTRL1_O`)
2. Inside of `CLK_BUFG_TOP` are 16 clusters of `BUFGCTRL_VPR`, and inside of `CLK_BUFG_BOT` are 16 clusters of `BUFGCTRL_VPR`. Each instance is given the correct FASM prefix (e.g. `BUFGCTRL_X0Y0`, `BUFGCTRL_X0Y1`, etc)
3. `CLK_BUFG_TOP` and `CLK_BUFG_BOT` are each instanced once inside the arch grid definition
4. Routing import should work as normal, because the top level port names match the expected wire names

litghost, 2019-09-14:
@acomodi You ask in IRC about potentially reasons a P&R BUFG/BUFH might not work.  This morning I remembered that the REBUF_BUFG cells behavior weirdly, where if there is an active signal on one side of the pip, the rebuf features need to be turned on.

I expect if you do a bit2fasm -> fasm2v -> Vivavo -> bit2fasm and diff (e.g. `<target>_vivado_diff_fasm`), you'll see that the REBUF_BUFG features were missing.

See the rebuf fuzzer for details: https://github.com/SymbiFlow/prjxray/blob/master/fuzzers/043-clk-rebuf-pips/generate.py#L57

acomodi, 2019-09-16:
@litghost I have made further experiments and these are my results:

1. As by you suggested, those REBUF_BUFG cells do appear in the FASM generated with bit2fasm after having used the generated verilog to produce a design through Vivado. This means that we could need to handle to handle these `fasm_features` as well. They look to something like this:
```
...
CLK_BUFG_REBUF_X60Y13.GCLK0_ENABLE_BELOW

CLK_BUFG_REBUF_X60Y38.CLK_BUFG_REBUF_R_CK_GCLK0_BOT.CLK_BUFG_REBUF_R_CK_GCLK0_TOP
CLK_BUFG_REBUF_X60Y38.GCLK0_ENABLE_ABOVE
CLK_BUFG_REBUF_X60Y38.GCLK0_ENABLE_BELOW
...
```
I still didn't understand what is the logic behind their activation.
`BELOW` and `ABOVE` seem inverted to me as the GCLK0 completely traverses the CLK_BUFG_REBUF_X60Y38 tile (so it makes sense that both of the REBUFs sides are active), but the CLK_BUFG_REBUF_X60Y13 (which is the most bottom REBUF tile) should not be traversed by GCLK0 (meaning that the signal has just to reach the BUFHCE tile and not going further for the test to work correctly).
The fact though is that I have tried to use the `top_bit.v.fasm` file, remove these REBUF features and tried to generate another bitstream with fasm2frames and they actually do not seem to be needed to have a working test.

2. The issue with the non-working implemented design most probably was the fact that I didn't add correct fasm features to the BUFGCTRL and BUFHCE pb_types:
```
E.g:
CLK_BUFG_BOT_R_X60Y48.BUFGCTRL.BUFGCTRL_X0Y0.IS_IGNORE1_INVERTED
CLK_BUFG_BOT_R_X60Y48.BUFGCTRL.BUFGCTRL_X0Y0.ZINV_CE0
CLK_BUFG_BOT_R_X60Y48.BUFGCTRL.BUFGCTRL_X0Y0.ZINV_S0
CLK_BUFG_BOT_R_X60Y48.BUFGCTRL.BUFGCTRL_X0Y0.ZPRESELECT_I0
```

I'll need to regenerate everything from the beginning (with these features added to the pb_type) and see if this solves the problem.

3. There is also another issue (but I think that is not a critical). FASM does also activate all the unused BUFGCTRLs and BUFHCE.

I will do some other tests, but I think we are close to have a working clock driven by BUFGs

litghost, 2019-09-16:
> 1. REBUF
> The fact though is that I have tried to use the `top_bit.v.fasm` file, remove these REBUF features and tried to generate another bitstream with fasm2frames and they actually do not seem to be needed to have a working test.

I believe those features have to do with clock quality, rather than a binary work/not work.  So for BUFG support to be complete, we should be emitting those features.

> 2. Inverted features

You don't need to emit the inversions, but you *do* need to ensure that the correct signals are reaching the blackbox (e.g. CE0 = true, CE1 = false, S0 = true, S1 = false, IGNORE0 = false, IGNORE1 = true).  In the BUFGCTRL, you *must* use the invert features to convert the default VCC signal into the correct signal.  If you do not, Vivado will complain on the fasm2v input.

See https://github.com/SymbiFlow/symbiflow-arch-defs/blob/master/xc7/techmap/cells_map.v#L989

> 3. Too BUFGCTRL's many in use

The way I've solved this in the SLICE is is to have each BUFGCTRL be a mode select with 2 items, empty and the BUFGCTRL black box.  By default the empty mode will emit no FASM features.

acomodi, 2019-09-16:
@litghost So, I am very close to the solution.

I have discovered that there are features of the `CLK_HROW_BOT_R` that need to be enabled. They are relative to the clock wire that is being used.

```
...
CLK_HROW_BOT_R_X60Y26.CLK_HROW_CK_IN_R0_ACTIVE
CLK_HROW_BOT_R_X60Y26.CLK_HROW_R_CK_GCLK0_ACTIVE
...
```

I discovered that by adding these two line above to a fasm generated through `counter_basys3_bin`, the test correctly works.

acomodi, 2019-09-17:
@litghost 
I have been able to implement the counter test which uses BUFGCTRL and BUFHCE and is working on HW. I had though to manually hack the .fasm by manually adding this line:
```
CLK_HROW_TOP_R_X60Y78.CLK_HROW_R_CK_GCLK0_ACTIVE
```

I am unsure on how to set these kind of fasm lines automatically.

I would say that the next steps will be the following:
- Understand how to handle the REBUF fasm features.
- As soon as we have break-out from the ROI, add support also for the `HCLK_CMT` and `HCLK_IOI` tiles. 
- Maybe also add the PLL support.
- Find a way to make the BUFHCE not placeable, as by you suggested. The main issue with a BUFHCE placement is that if a tile is placed in a different clock region, it will become unroutable from the clock point of view.

litghost, 2019-09-17:
> I am unsure on how to set these kind of fasm lines automatically.

https://github.com/SymbiFlow/symbiflow-arch-defs/blob/master/xc7/utils/prjxray_routing_import.py#L83 can be use for attaching features that have a 1-to-1 pip to feature relationship.

acomodi, 2019-09-17:
@litghost Great. So basically I will need to add the regexp relative to the pip that is being used and add the fasm line (as it happens for `CASCOOUT` for instance).

I have stumbled upon an issue during `channel.db` creation after a rebase:
```
2019-09-17 18:50:13.874983 Annotate pin feeds
Dumping database to '/home/alessandro/projects/symbiflow/symbiflow-arch-defs/build/xc7/archs/artix7/devices/xc7a50t-basys3-roi-virt/channels.db'
100% (260212 of 260212) |############################################################################| Elapsed Time: 0:00:06 Time:  0:00:06
Traceback (most recent call last):
  File "/home/alessandro/projects/symbiflow/symbiflow-arch-defs/xc7/utils/prjxray_create_edges.py", line 1805, in <module>
    main()
  File "/home/alessandro/projects/symbiflow/symbiflow-arch-defs/xc7/utils/prjxray_create_edges.py", line 1788, in main
    annotate_pin_feeds(conn)
  File "/home/alessandro/projects/symbiflow/symbiflow-arch-defs/xc7/utils/prjxray_create_edges.py", line 1460, in annotate_pin_feeds
    tracks=list(),
  File "/home/alessandro/projects/symbiflow/symbiflow-arch-defs/xc7/utils/prjxray_create_edges.py", line 1386, in walk_and_mark_segment
    pin_graph_node_pkey, tracks
  File "/home/alessandro/projects/symbiflow/symbiflow-arch-defs/xc7/utils/prjxray_create_edges.py", line 1386, in walk_and_mark_segment
    pin_graph_node_pkey, tracks
  File "/home/alessandro/projects/symbiflow/symbiflow-arch-defs/xc7/utils/prjxray_create_edges.py", line 1397, in walk_and_mark_segment
    tracks=tracks
  File "/home/alessandro/projects/symbiflow/symbiflow-arch-defs/xc7/utils/prjxray_create_edges.py", line 1265, in set_pin_connection
    assert len(source_wires) <= 1
AssertionError
```

The parts of code that produce this assertion error have been added with [this commit](https://github.com/SymbiFlow/symbiflow-arch-defs/commit/49812c3ec88e09d1c71eadf45381514a446f8f56). Any idea on what could be cause this failure?

litghost, 2019-09-17:
> The parts of code that produce this assertion error have been added with [this commit](https://github.com/SymbiFlow/symbiflow-arch-defs/commit/49812c3ec88e09d1c71eadf45381514a446f8f56). Any idea on what could be cause this failure?

It's trying to annotate the connection box for an IPIN/OPIN (e.g. the connection to the routing interface) for, but it found an ambiguous case.  You'll need to see which wire it is referring too.  It is possible that the connection box logic needs some tweaking for the clock network.

acomodi, 2019-09-18:
@litghost Ok. As far as I could understand, the issue is indeed in the clocking tree. The fact is that there are many possible lookahead pin nodes for a node belonging to the `CLK_HROW_TOP_R` tile. 
Now, the problem is that the clock network going out from the BUFHCE directly connects to all the logic tiles of each clock region.

I was thinking to skip, if possible, the creation of the lookahead for the clock tree. The problem with this is understanding if the wire is indeed relative to the clock network.

Moreover, I have tried to temporarily comment the assetion and got to another issue: 
```
Traceback (most recent call last):
  File "/home/alessandro/projects/symbiflow/symbiflow-arch-defs/xc7/utils/prjxray_create_edges.py", line 1846, in <module>
    main()
  File "/home/alessandro/projects/symbiflow/symbiflow-arch-defs/xc7/utils/prjxray_create_edges.py", line 1829, in main
    annotate_pin_feeds(conn)
  File "/home/alessandro/projects/symbiflow/symbiflow-arch-defs/xc7/utils/prjxray_create_edges.py", line 1522, in annotate_pin_feeds
    tracks=list(),
  File "/home/alessandro/projects/symbiflow/symbiflow-arch-defs/xc7/utils/prjxray_create_edges.py", line 1427, in walk_and_mark_segment
    pin_graph_node_pkey, tracks
  File "/home/alessandro/projects/symbiflow/symbiflow-arch-defs/xc7/utils/prjxray_create_edges.py", line 1427, in walk_and_mark_segment
    pin_graph_node_pkey, tracks
  File "/home/alessandro/projects/symbiflow/symbiflow-arch-defs/xc7/utils/prjxray_create_edges.py", line 1427, in walk_and_mark_segment
    pin_graph_node_pkey, tracks
  [Previous line repeated 990 more times]
  File "/home/alessandro/projects/symbiflow/symbiflow-arch-defs/xc7/utils/prjxray_create_edges.py", line 1335, in walk_and_mark_segment
    graph_node_type = NodeType(cur.fetchone()[0])
  File "/home/alessandro/projects/symbiflow/symbiflow-arch-defs/build/env/conda/lib/python3.7/enum.py", line 310, in __call__
    return cls.__new__(cls, value)
  File "/home/alessandro/projects/symbiflow/symbiflow-arch-defs/build/env/conda/lib/python3.7/enum.py", line 530, in __new__
    if type(value) is cls:
RecursionError: maximum recursion depth exceeded while calling a Python object
xc7/archs/artix7/devices/xc7a50t-basys3-roi-virt/CMakeFiles/file_xc7_archs_artix7_devices_xc7a50t-basys3-roi-virt_channels.db.dir/build.make:66: recipe for target 'xc7/archs/artix7/devices/xc7a50t-basys3-roi-virt/channels.db' failed
```

Meaning that `walk_and_mark_segment` never gets to an end and is indefinitely called recursively.

litghost, 2019-09-18:
> I was thinking to skip, if possible, the creation of the lookahead for the clock tree. The problem with this is understanding if the wire is indeed relative to the clock network.

The was to do this is probably to blacklist clock sources and sinks (think `BUFGCTRL.I[01]`, `BUFGTRL.O`, `BUFHCE.[IO]`, etc) from `walk_and_mark_segment `, as `walk_and_mark_segment` and is really only correct for things that touch the general interconnect.  In the future, some form of clock network annotation is needed for the VPR 2-stage routing, and having the list of sinks/sources that belong to the dedicate clock network will likely be required, so the work isn't wasteful.

acomodi, 2019-09-18:
@litghost All right. For the time being I think I will blacklist the clock source/sink than, just to arrive at further stages of the build.
I'll open an issue about that, to keep track of the fact that we need to treat the clock network source/sinks annotations as well.

closed 2019-12-12 by litghost