f4pga/f4pga-arch-defs #427: vtr from conda does not recognize --round_robin_prepacking option

kgugala, 2019-03-05:
this option has been added by the https://github.com/SymbiFlow/symbiflow-arch-defs/commit/60700b81aa5221ea99c72100aa26c3e67adb28b3 commit.

The option is implemented in vtr. We just need to bump the conda package

kgugala, 2019-03-05:
also genfasm should be updated:
```

Unexpected command-line argument '--round_robin_prepacking'

usage: genfasm architecture circuit [--pack] [--place] [--route] 
       [--analysis] [--disp {on, off}] [-h] [--version] [--device DEVICE_NAME] 
       [-j NUM_WORKERS] [--timing_analysis {on, off}] 
       [--route_chan_width CHANNEL_WIDTH] [OTHER_OPTIONS ...]

```

mithro, 2019-03-05:
@kgugala Why would the `--round_robin_prepacking` be passed to the `genfasm` command?

kgugala, 2019-03-05:
this is because all the vtr tools share the same arguments. The arguments are defined here: https://github.com/SymbiFlow/symbiflow-arch-defs/blob/master/xc7/make/arch_define.cmake#L27

mithro, 2019-03-05:
To rebuild vpr follow https://github.com/SymbiFlow/conda-packages/pull/10

closed 2019-03-05 by mithro