f4pga/f4pga-arch-defs #438: disable RAMB36 in Yosys

acomodi, 2019-03-06:
To have picosoc running the program data is saved in BRAMs. To allow it to correctly compile right now the bram yosys techlibs (`brams.txt`) has to be updated as follows:

```
#match $__XILINX_RAMB36_SDP
#  min bits 4096
#  min efficiency 5
#  shuffle_enable B
#  make_transp
#  or_next_if_better
#endmatch

match $__XILINX_RAMB18_SDP
  min bits 4096
  min efficiency 5
  shuffle_enable B
  make_transp
  or_next_if_better
endmatch

#match $__XILINX_RAMB36_TDP
#  min bits 4096
#  min efficiency 5
#  shuffle_enable B
#  make_transp
#  or_next_if_better
#endmatch

match $__XILINX_RAMB18_TDP
  min bits 4096
  min efficiency 5
  shuffle_enable B
  make_transp
endmatch
```

This is to exclude RAMB36

litghost, 2019-03-06:
Disabling RAMB36 inference is a good short term solution, but we should enable RAMB36 place and route in symbiflow-arch-defs as well.

acomodi, 2019-03-06:
Agreed. For now, when compiling picosoc yosys has to be overriden with the local version that has RAMB36 disabled.

litghost, 2019-09-13:
Closing this issue and tracking in #902

closed 2019-09-13 by litghost