f4pga/f4pga-arch-defs #463: Revisit python dependencies and requirements file

elmsfu, 2019-03-14:
We essentially have python package dependencies scoped in the cmake files, but also have a requirements file at the top level. We should only have a single method of tracking dependencies.

Do we need to scope any packages and have them installed only if you are building a target that depends on them?

litghost, 2019-03-14:
No, but there does need to be some link between dependency installation and targets that need them.

elmsfu, 2019-03-14:
We have several libraries and several executable modules. One idea is to wrap these up as packages with a setup.py so they can be installed in the env and tested independently. The granularity of dependencies would be less, but easier to describe in CMake (I think). For example, if a build needs something from xc7/utils you will probably need them all so no harm in that level of dependency.

mithro, 2019-03-14:
@elmsfu Which libraries / executables are you talking about?

I agree that we should probably revisit requirements.txt verse inside cmake files verse setup.py, etc. 

I definitely think the [third_party/prjxray](third_party/prjxray) ones should become proper Python modules (with setup.py and such). Unsure about the local stuff in this repo.

elmsfu, 2019-03-14:
These are top of mind:
 * [utils](https://github.com/SymbiFlow/symbiflow-arch-defs/tree/daee11d750e6ba6948f0c398690abd4dbb301189/utils)
 * [utils/lib](https://github.com/SymbiFlow/symbiflow-arch-defs/tree/daee11d750e6ba6948f0c398690abd4dbb301189/utils/lib)
 * [ice40/utils](https://github.com/SymbiFlow/symbiflow-arch-defs/tree/daee11d750e6ba6948f0c398690abd4dbb301189/ice40/utils)
 * [xc7/utils](https://github.com/SymbiFlow/symbiflow-arch-defs/tree/daee11d750e6ba6948f0c398690abd4dbb301189/xc7/utils)
 * [utils/vlog/yosys](https://github.com/SymbiFlow/symbiflow-arch-defs/tree/daee11d750e6ba6948f0c398690abd4dbb301189/utils/vlog/yosys)

mithro, 2019-03-15:
Just going through those directories;

 * `xc7/utils` are all command line tools
 * `ice40/utils` is mostly command line tools, with one library file `pcf.py`
 * `utils/*.py` seem to be mostly command line tools. Seems to be one library file `eblif.py`.
 * `utils/lib` these actually seem to be library / support functions with `utils/lib/rr_graph` and `utils/lib/rr_graph_xml` been strongly libraries.
 * `utils/vlog` seems to be fairly self contained with a couple of command line tools and library support functions in `utils/vlog/yosys`

elmsfu, 2019-04-02:
overlaps #279

mithro, 2019-04-02:
Just FYI - it is worth also thinking about setup.py verse requirements.txt

 * https://packaging.python.org/discussions/install-requires-vs-requirements/
 * https://stackoverflow.com/questions/43658870/requirements-txt-vs-setup-py
 * https://caremad.io/posts/2013/07/setup-vs-requirement/

mithro, 2019-05-04:
I was thinking more about this today. There are a couple of things we have;

 * Library for working with VtR rr_graph files.
   - [utils/lib/rr_graph](https://github.com/SymbiFlow/symbiflow-arch-defs/tree/master/utils/lib/rr_graph)
   - [utils/lib/rr_graph_xml](https://github.com/SymbiFlow/symbiflow-arch-defs/tree/master/utils/lib/rr_graph_xml)
   - [utils/check_graph.py](https://github.com/SymbiFlow/symbiflow-arch-defs/blob/master/utils/check_graph.py)
   - [utils/print_graph.py](https://github.com/SymbiFlow/symbiflow-arch-defs/blob/master/utils/print_graph.py)
   - [utils/reparse_graph.py](https://github.com/SymbiFlow/symbiflow-arch-defs/blob/master/utils/print_graph.py)
   - etc...

 * The `v2x` tool + library.
   - Pretty much self contained under [utils/vlog](https://github.com/SymbiFlow/symbiflow-arch-defs/tree/master/utils/vlog)

 * Tools around finding includes
   - [utils/deps_verilog.py](https://github.com/SymbiFlow/symbiflow-arch-defs/blob/master/utils/deps_verilog.py)
   - [utils/deps_xml.py](https://github.com/SymbiFlow/symbiflow-arch-defs/blob/master/utils/deps_xml.py)

 * Tools around generation of common parts...
   - [utils/mux_gen.py](https://github.com/SymbiFlow/symbiflow-arch-defs/blob/master/utils/mux_gen.py)
   - (soon) [utils/generate_latches.py](https://wavedrom.com/tutorial.html).

litghost, 2020-10-29:
This issue has mostly been resolved with the move to conda environment files.

closed 2020-10-29 by litghost