f4pga/f4pga-arch-defs #464: ice40: Implement ColBuf in routing import

elmsfu, 2019-03-14:
Column Buffer enabling is inferred via HLC. Moving to FASM with #447 turns on all the Column Buffers for now. This works but wastes power.

I think the correct way would be to model them in the routing graph and emit features when used. It could also be done in a post-PnR (place and route) fixup step.

litghost, 2019-03-14:
> I think the correct way would be to model them in the routing graph and emit features when used. It could also be done in a post-PnR (place and route) fixup step.

Definitely a post-PnR fixup.  Clock networks in 7-series have something similar, but it is definitely as if A then B thing.  Sounds like the Column Buffers are similiar?

elmsfu, 2019-03-14:
> > I think the correct way would be to model them in the routing graph and emit features when used. It could also be done in a post-PnR (place and route) fixup step.
> 
> Definitely a post-PnR fixup. Clock networks in 7-series have something similar, but it is definitely an if A then B thing. Sounds like the Column Buffers are similiar?

Can you explain the trade-offs (downside) of modeling in the routing graph?

litghost, 2019-03-14:
> Can you explain the trade-offs (downside) of modeling in the routing graph?

I actually think the 7-series BUF REBUF case can be handled in the routing graph, so I take back my previous statement.  I am worried about some of the feature enables for the clock being done in the routing graph, and it may be simpler to test and code in a post-PnR solution.  I guess the distinction is if the feature can be tied to one edge, then maybe it will work out.