f4pga/f4pga-arch-defs #481: Create build target for to run python tests

elmsfu, 2019-03-20:
Currently existing python tests are not run. There should be a top level target to run all python tests.

Travis script should then call that target.

This can be addressed after #463

elmsfu, 2019-03-28:
Looks like we have an attempt to support `pytest`, but previously I think we had `doctest` running. More recently we have used `unittest`.

The target should run `unittest` and `doctest`. If we use `pytest` we should phase them out in favor of `unittest`

mithro, 2019-03-28:
@elmsfu pytest is a "test runner". It runs [`unittest`](https://docs.pytest.org/en/latest/unittest.html) and [`doctest`](https://docs.pytest.org/en/latest/doctest.html) tests.

We should continue to use `unittest` and `doctest` but use pytest as the test runner when running everything at once.

elmsfu, 2019-04-03:
Fixed by #520

closed 2019-04-03 by elmsfu

mithro, 2019-04-03:
Yay! \o/