f4pga/f4pga-arch-defs #482: Error in prjxray import scripts don't get shown

mithro, 2019-03-20:
If you have an error in the prjxray import scripts it just hangs at a progress bar that looks like the following;
```
- |                                                                                                   #                                                                                  | 0 Elapsed Time: 0:17:09
```

When you ctrl-c the command you get the actual error which was;
```
Traceback (most recent call last):
  File "/usr/local/google/home/tansell/github/SymbiFlow/symbiflow-arch-defs-clean/xc7/utils/prjxray_form_channels.py", line 709, in main
Traceback (most recent call last):
  File "/usr/local/google/home/tansell/github/SymbiFlow/symbiflow-arch-defs-clean/xc7/utils/prjxray_form_channels.py", line 709, in main
    import_grid(db, grid, conn)
    import_grid(db, grid, conn)
  File "/usr/local/google/home/tansell/github/SymbiFlow/symbiflow-arch-defs-clean/xc7/utils/prjxray_form_channels.py", line 163, in import_grid
  File "/usr/local/google/home/tansell/github/SymbiFlow/symbiflow-arch-defs-clean/xc7/utils/prjxray_form_channels.py", line 163, in import_grid
    import_tile_type(db, c, tile_types, site_types, gridinfo.tile_type)
  File "/usr/local/google/home/tansell/github/SymbiFlow/symbiflow-arch-defs-clean/xc7/utils/prjxray_form_channels.py", line 99, in import_tile_type
    import_site_type(db, c, site_types, site.type)
    import_tile_type(db, c, tile_types, site_types, gridinfo.tile_type)
  File "/usr/local/google/home/tansell/github/SymbiFlow/symbiflow-arch-defs-clean/xc7/utils/prjxray_form_channels.py", line 99, in import_tile_type
    import_site_type(db, c, site_types, site.type)
  File "/usr/local/google/home/tansell/github/SymbiFlow/symbiflow-arch-defs-clean/xc7/utils/prjxray_form_channels.py", line 50, in import_site_type
    site_type = db.get_site_type(site_type_name)
  File "/usr/local/google/home/tansell/github/SymbiFlow/symbiflow-arch-defs-clean/third_party/prjxray/prjxray/db.py", line 142, in get_site_type
    return site_type.SiteType(site_type_data)
  File "/usr/local/google/home/tansell/github/SymbiFlow/symbiflow-arch-defs-clean/xc7/utils/prjxray_form_channels.py", line 50, in import_site_type
    site_type = db.get_site_type(site_type_name)
  File "/usr/local/google/home/tansell/github/SymbiFlow/symbiflow-arch-defs-clean/third_party/prjxray/prjxray/db.py", line 142, in get_site_type
    return site_type.SiteType(site_type_data)
  File "/usr/local/google/home/tansell/github/SymbiFlow/symbiflow-arch-defs-clean/third_party/prjxray/prjxray/site_type.py", line 22, in __init__
    direction=SitePinDirection(site_pin_info['direction']),
  File "/usr/local/google/home/tansell/github/SymbiFlow/symbiflow-arch-defs-clean/build/env/conda/lib/python3.7/enum.py", line 309, in __call__
    return cls.__new__(cls, value)
  File "/usr/local/google/home/tansell/github/SymbiFlow/symbiflow-arch-defs-clean/build/env/conda/lib/python3.7/enum.py", line 543, in __new__
    return cls._missing_(value)
  File "/usr/local/google/home/tansell/github/SymbiFlow/symbiflow-arch-defs-clean/build/env/conda/lib/python3.7/enum.py", line 556, in _missing_
    raise ValueError("%r is not a valid %s" % (value, cls.__name__))
ValueError: 'INOUT' is not a valid SitePinDirection
```

The KeyboardInterrupt seems to have been received by;
```

Traceback (most recent call last):
  File "/usr/local/google/home/tansell/github/SymbiFlow/symbiflow-arch-defs-clean/xc7/utils/prjxray_form_channels.py", line 723, in <module>
    main()
  File "/usr/local/google/home/tansell/github/SymbiFlow/symbiflow-arch-defs-clean/xc7/utils/prjxray_form_channels.py", line 720, in main
    print('{} Flushing database back to file "{}"'.format(datetime.datetime.now(), args.connection_database))
  File "/usr/local/google/home/tansell/github/SymbiFlow/symbiflow-arch-defs-clean/xc7/utils/prjxray_db_cache.py", line 54, in __exit__
    self.memory_connection.backup(self.file_connection, pages=100, progress=self._progress)
  File "/usr/local/google/home/tansell/github/SymbiFlow/symbiflow-arch-defs-clean/xc7/utils/prjxray_db_cache.py", line 63, in _progress
    def _progress(self, status, remaining, total):
  File "/usr/local/google/home/tansell/github/SymbiFlow/symbiflow-arch-defs-clean/xc7/utils/prjxray_db_cache.py", line 63, in _progress
    def _progress(self, status, remaining, total):
KeyboardInterrupt
KeyboardInterrupt
```

mithro, 2019-03-21:
I believe this has something to do with the `__exit__` stuff added?

mithro, 2019-03-21:
@litghost Thoughts?

litghost, 2019-03-27:
This is caused by memory_connection.in_transaction being true.  I encountered this bug in another context, where I had forgot to call conn.commit() before the cache attempted to flush.  I believe I understand the fix.

PR #512 (2019-03-28): Fix database cache for in progress transactions.
If the database cache __exit__ method is called when a transaction
is in progress, it will never proceed.  Handle the exception case by
rolling back, and generate an assertion error in the nominal path.

Fixes #482.
commit e0c633eb (2019-03-28): Fix database cache for in progress transactions.

If the database cache __exit__ method is called when a transaction
is in progress, it will never proceed.  Handle the exception case by
rolling back, and generate an assertion error in the nominal path.

Fixes #482.
closed 2019-03-28 by litghost