f4pga/f4pga-arch-defs #491: [XC7] Some designs fail on hardware

litghost, 2019-03-22:
@mkurc came up with https://github.com/SymbiFlow/symbiflow-arch-defs/pull/490 that works at small N and fails at larger N.

This issue tracks the investigation into why this is happening and potential solutions.

litghost, 2019-03-22:
I've seen this issue when I connected all IO lines to the xc7/tests/bram_test.  There was a known issue where SHORT's should be used instead of MUX's on channel connections, but that did not appear to fix the issue.  Something else is going on.

litghost, 2019-03-22:
The changes in https://github.com/SymbiFlow/symbiflow-arch-defs/pull/494 cause the BRAM tests to start failing.  I suspect something is happening at higher router pressure that is cause the failure.

litghost, 2019-03-22:
So #494 causes the DRAM test to break too, which makes me believe this might be an ROI bug rather than something else.

litghost, 2019-03-22:
It looks like the disabling round robin packing helps with the failure in #494.  I'll test if it helps with #490.

litghost, 2019-03-22:
I think this is related to setting the DLUT.RAM bit and then not having an DRAM's connected.  This was caused by the round robin packing because it would have the DRAM mode even if it wasn't required.  This resulted in the DLUT.RAM bit being set, but no actual DRAM's being in use.

litghost, 2019-03-29:
@mkurc-ant I'm having trouble getting any of the 9-scalable_proc tests to work.  Can you provide a commit in which 9-scalable_proc works on hardware?

litghost, 2019-03-29:
@mkurc-ant: I figured out what was going on, the RST sw was in reset

litghost, 2019-03-29:
@mkurc-ant: Looks like BRAM N1 passes, but DRAM N1 fails.  Does this match your results?

mkurc-ant, 2019-03-29:
@litghost: For N=1 and 2 it works correctly for me. Here are my results:

symbiflow-arch-defs: da63099
Yosys: 87846729 (built locally)
VPR: f725c5ac0 (built locally)

Round robin prepacking: ON
N=1 - Working correctly
N=2 - Working correctly
N=3 - Working, computation errors
N=4 - Working, computation errors
N=5 - Working, computation errors
N=6 - Working, computation errors
N=7 - Working, computation errors (every second or third word)
N=8 - Working, computation errors (every second or third word)

Round robin prepacking: OFF
N=1 - Working correctly
N=2 - Computation errors and some ASCII chars are messed up
N=3 - Receiving unexpected ASCII chars (which suppose to belong to [0-9][A-F]).
N=4 - Same as for N=3
N=5 - Working, computation errors. ASCII chars are ok
N=6 - ASCII transmission totally messed up. Unexpected chars, no new lines
N=7 - **Working flawlessly (!)**
N=8 - Failed to place. Since 15th iteration there have been 2-5 overused routes. I have waited until 35th and then cancelled.

And again when using Yosys+Vivado everything works fine (checked up to N=8)

I've also noticed that routing with round-robin prepacking off takes more time (didn't measured).

But the most surprising is that with RR off and N=7 the design works.

acomodi, 2019-03-29:
@mkurc-ant @litghost Would it be worth to run the same tests with the version of VTR before the upstream merge and see if there are issues with the merge? (I am referring to the first merge https://github.com/SymbiFlow/vtr-verilog-to-routing/pull/20)

litghost, 2019-03-29:
> @litghost: For N=1 and 2 it works correctly for me.

Are you testing with BRAM or DRAM?  I see that N = 1 works for BRAM, but I'm seeing N = 1 fail for DRAM.

litghost, 2019-03-29:
> @mkurc-ant @litghost Would it be worth to run the same tests with the version of VTR before the upstream merge and see if there are issues with the merge? (I am referring to the first merge [SymbiFlow/vtr-verilog-to-routing#20](https://github.com/SymbiFlow/vtr-verilog-to-routing/pull/20))

Naw, I have a feeling none of this is truely a VPR bug, but an arch-def bug of some kind.

mkurc-ant, 2019-03-29:
@litghost These results I've posted are for DRAM

litghost, 2019-03-29:
> @litghost These results I've posted are for DRAM

Thanks for clarifying.  My focus today will be to get some bugs worked out of fasm2bels on the scalable test and see if I can identify the issue.

litghost, 2019-04-11:
I've replicated the scalable poc failure using fasm2verilog and Vivado simulation.  I've gotten far enough to determine that the fsm_pulse_cnt isn't incrementing, will debug further tomorrow.

litghost, 2019-04-13:
Final discovered the bug, it was in the FASM output for the routing graph, and it affected HLC and FASM generation for both xc7 and ice40.

commit 50ec1ac8 (2019-04-16): Merge pull request #596 from litghost/update_vtr

Update VTR which fixes #491
commit 84fb1110 (2019-04-16): Update VTR which fixes #491
closed 2019-04-16 by litghost