f4pga/f4pga-arch-defs #524: Always PPIPs should be modelled as shorts

litghost, 2019-04-02:


litghost, 2019-04-02:
FANx/BYPx usage in CLB tiles:

```
CLB:
FAN0: X0_AI
FAN1: 
FAN2: X0_BI
FAN3: X0_DI
FAN4: X0_WE
FAN5: X0_CI
FAN6: X1_CE
FAN7: X0_CE
BYP0: X1_AX
BYP1: X0_AX
BYP2: X1_CX
BYP3: X0_CX
BYP4: X1_BX
BYP5: X0_BX
BYP6: X1_DX
BYP7: X0_DX
```

Given how FANx or BYPx signals are used, I'm not sure how modelling these edges as shorts should make a difference.  If the signal is unused, then it toggling should be safe to ignore.  The exception is the SR and CE signals, which have the SRUSEDMUX and CEUSEDMUX bits to protect CLB from stray signals.

mithro, 2019-04-02:
Still, from a semantic view - it would be good for them to be shorts.

litghost, 2019-04-02:
> Still, from a semantic view - it would be good for them to be shorts.

It causes routing output to break, because VPR sees an active net connected to a block IPIN that is not part of the netlist.  I'm not sure how to solve that right now.