f4pga/f4pga-arch-defs #539: Define and implement reasonable all target

litghost, 2019-04-05:
Current default all target as of c66597570807a07734e59a98fe0456b2b2a39ff0 is to build every bitstream.  This is almost certainly not the right thing to do as the default all target.  This issue about defining a reasonable all target.

elmsfu, 2019-04-05:
I agree that all bitstreams is probably not correct. Maybe "small" tests would be a good goal with the aim to cover of all the blocks and modes. But that leaves any stress test, but not sure if travis is a good match for that.

litghost, 2019-04-05:
I think it problably time to get the kokoro instances going to symbiflow, because they are beefier machines, and will be able to run @ -j16 and get results in a reasonable amount of time.

closed 2019-12-12 by litghost