f4pga/f4pga-arch-defs #549: [XC7] RAM128X1D test is no longer working

litghost, 2019-04-08:


HackerFoo, 2019-04-22:
```
Results from re-enabling the dram_1_128x1d FPGA target and running _bin:

# Packing
Begin packing '/home/dusty/src/symbiflow-arch-defs/build/xc7/tests/dram/dram_1_128x1d/artix7-xc7a50t-basys3-roi-virt-xc7a50t-basys3-test/top.eblif'.

After removing unused inputs...
	total blocks: 26, total nets: 25, total inputs: 17, total outputs: 2
Begin prepacking.
Finish prepacking.
Using inter-cluster delay: 6.74304e-10
Packing with pin utilization targets: BLK_TI-CLBLL_R:0.8,1 BLK_TI-CLBLL_L:1,1 BLK_TI-CLBLM_R:1,1 BLK_TI-CLBLM_L:1,1 BLK_TI-BRAM_L:1,1 BLK_SY-INPAD:1,1 BLK_SY-OUTPAD:1,1 BLK_SY-VCC:1,1 BLK_SY-GND:1,1
Not enough resources expand FPGA size to (116 x 158)
Complex block 0: ram0.ram3, type: BLK_TI-CLBLM_R ....
Complex block 1: ram0.f7a_mux, type: BLK_TI-CLBLM_L .Error 1: 
Type: Packing
File: /build/source/vpr/src/pack/cluster.cpp
Line: 1976
Message: Can not find any logic block that can implement molecule.
	Pattern DRAM128_DP ram0.f7a_mux

Routing was impossible!
Net 'ram0.blut_o6' is impossible to route within proposed BLK_TI-CLBLM_L cluster
Failed route at end, repack cluster trying detailed routing at each stage.
Routing was impossible!
Net 'ram0.blut_o6' is impossible to route within proposed BLK_TI-CLBLM_L cluster
Routing was impossible!
Net 'ram0.blut_o6' is impossible to route within proposed BLK_TI-CLBLM_R cluster

```

mithro, 2019-04-22:
https://github.com/SymbiFlow/symbiflow-arch-defs/blob/6850f2c665a8fabb12a771067d6d2a2d7ea266c0/xc7/primitives/slicem/slicem.pb_type.xml#L569-L574

mithro, 2019-04-22:
You probably want to enable the echo files and/or enable [this define](https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vpr/src/pack/cluster_router.cpp#L37).

mithro, 2019-04-22:
It might also be helpful to look at https://github.com/SymbiFlow/symbiflow-arch-defs/tree/master/xc7/primitives/slicem/Ndram

HackerFoo, 2019-04-22:
Results from _route_echo:
```
# Packing
Begin packing '/home/dusty/src/symbiflow-arch-defs/build/xc7/tests/dram/dram_1_128x1d/artix7-xc7a50t-basys3-roi-virt-xc7a50t-basys3-test/top.eblif'.

After removing unused inputs...
	total blocks: 26, total nets: 25, total inputs: 17, total outputs: 2
Begin prepacking.
Finish prepacking.
Using inter-cluster delay: 6.74304e-10
Packing with pin utilization targets: BLK_TI-CLBLL_R:0.8,1 BLK_TI-CLBLL_L:1,1 BLK_TI-CLBLM_R:1,1 BLK_TI-CLBLM_L:1,1 BLK_TI-BRAM_L:1,1 BLK_SY-INPAD:1,1 BLK_SY-OUTPAD:1,1 BLK_SY-VCC:1,1 BLK_SY-GND:1,1
Complex block 0:
	Seed: 'ram0.ram3' (DPRAM128) 
	PASSED_SEED: Block Type BLK_TI-CLBLM_R
Not enough resources expand FPGA size to (116 x 158)
	FAILED_CHECK: ram0.f7a_mux type MUXF7 check 1molecule DRAM128_DP molecule_size 3
	PASSED: ram0.ram1 type DPRAM128
	FAILED_CHECK: ram0.f7a_mux type MUXF7 check 1molecule DRAM128_DP molecule_size 3
	PASSED: ram0.ram2 type DPRAM128
	FAILED_CHECK: ram0.f7a_mux type MUXF7 check 1molecule DRAM128_DP molecule_size 3
	PASSED: ram0.ram0 type SPRAM128
	FAILED_CHECK: ram0.f7a_mux type MUXF7 check 1molecule DRAM128_DP molecule_size 3
	Found unrelated molecule to cluster
	Passed route at end.
Complex block 1:
	Seed: 'ram0.f7a_mux' (MUXF7) molecule_type DRAM128_DP molecule_size 3
	FAILED_SEED: Block Type BLK_TI-CLBLL_R
	FAILED_SEED: Block Type BLK_TI-CLBLL_L
	PASSED_SEED: Block Type BLK_TI-CLBLM_L
Error 1: 
Type: Packing
File: /tmp/really-really-really-really-really-really-really-really-really-really-really-really-really-long-path/conda/conda-bld/vtr_1555613142362/work/vpr/src/pack/cluster.cpp
Line: 1976
Message: Can not find any logic block that can implement molecule.
	Pattern DRAM128_DP ram0.f7a_mux

Routing was impossible!
Net 'ram0.blut_o6' is impossible to route within proposed BLK_TI-CLBLM_L cluster
Failed route at end, repack cluster trying detailed routing at each stage.
Complex block 1:
	Seed: 'ram0.f7a_mux' (MUXF7) molecule_type DRAM128_DP molecule_size 3
	FAILED_SEED: Block Type BLK_TI-CLBLL_R
	FAILED_SEED: Block Type BLK_TI-CLBLL_L
Routing was impossible!
Net 'ram0.blut_o6' is impossible to route within proposed BLK_TI-CLBLM_L cluster
	FAILED_SEED: Block Type BLK_TI-CLBLM_L
Routing was impossible!
Net 'ram0.blut_o6' is impossible to route within proposed BLK_TI-CLBLM_R cluster
	FAILED_SEED: Block Type BLK_TI-CLBLM_R
# Packing took 0.05 seconds (max_rss 55.2 MiB, delta_rss +40.1 MiB)
The entire flow of VPR took 0.16 seconds (max_rss 55.2 MiB)
make[3]: *** [xc7/tests/dram/CMakeFiles/dram_1_128x1d_route_echo.dir/build.make:104: xc7/tests/dram/dram_1_128x1d/artix7-xc7a50t-basys3-roi-virt-xc7a50t-basys3-test/echo/top.net] Error 1
make[2]: *** [CMakeFiles/Makefile2:46514: xc7/tests/dram/CMakeFiles/dram_1_128x1d_route_echo.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:46521: xc7/tests/dram/CMakeFiles/dram_1_128x1d_route_echo.dir/rule] Error 2
make: *** [Makefile:14488: dram_1_128x1d_route_echo] Error 2

```

mithro, 2019-04-22:
There should be a prepack file which has the molecules in it...

mithro, 2019-04-22:
I think that pack-pattern is suppose to join two 128 bit LUT rams together into a molecule so that WAUSED is correctly set...

mithro, 2019-04-22:
@HackerFoo What does the molecule echo file show?

litghost, 2019-05-03:
Fixed by #663

closed 2019-05-03 by litghost