f4pga/f4pga-arch-defs #559: Rely on external package management as much as possible

HackerFoo, 2019-04-08:
Automatically loading dependencies via Conda and PIP causes difficulty when trying to integrate with the system package manager, or when working offline.

In particular, Nix (and hence NixOS) controls dependencies to ensure repeatable builds, so it would be better to allow Nix to handle the dependencies.

The apparent approach would be to use `cmake -DUSE_CONDA=FALSE ..`, but this seems to be currently broken.

mithro, 2019-04-08:
@HackerFoo - External package managers (IE Debian / Ubuntu apt, Fedora, etc) don't have new enough versions or our custom patched versions. The other option would be to compile from scratch.

conda gives us a way to distribute (rather than compile from scratch) binary versions of our tooling.

litghost, 2019-04-08:
Conda setup should only occur the first time you build, much like a system package manager.  If conda is getting invoked after initial setup, that is likely a bug.

HackerFoo, 2019-04-09:
This works okay for development, but may cause problems when deploying to e.g. test servers. It's best to leave the packaging to the package managers, and either distribute packages directly or set up repositories (e.g. PPAs for Ubuntu.) Another option is to use Nix, or use a separate script that can be run before building.

I've also seen a single build download Miniconda at least 6 times, even if it has just been run. I think it was `buttons_basys3_prog`.

litghost, 2019-04-09:
> I've also seen a single build download Miniconda at least 6 times, even if it has just been run. I think it was `buttons_basys3_prog`.

Then that's simply a bug.

elmsfu, 2019-04-09:
I would have written a [shorter reply if I had the time](https://en.wikiquote.org/wiki/Blaise_Pascal#Pascal_plus_longue)

I think there is room to improve around this area, but we are still not close to stable and rely on integration of several repos that are in flux. I don't think this is an area we should attack yet. There is probably a bigger question for deploying for non-development.

## My understanding of current (2019-04-10) state
Currently we have 4 methods that I know of for external dependencies (or at least externally based if we have a custom fork).

1. conda (vpr, yosys)
1. submodule (icestorm, netlistsvg, fasm)
1. pip (should probably wrap all python as installable modules and have finer grain deps)
1. requirements.txt (not used by build system by default)

My experience has been that most projects choose to list dependencies in documentation and maybe provide a script to run for development/ease-of-use for popular OS/distro. CI can then run that script as part of setup.

## Potential options

1. Status Quo
Keep what we have as it's working. Although we should revisit VPR as an optional submodule for developers.

1. Supermodule/superproject
Switch from parent module (lots of code and dependencies) to superproject (a repo that only has a few scripts and then several submodules)
Anything that is in flux would be a submodule and built as part of the overall build. But this makes encapsulation of theenvironment a little harder.

1. Docker
Yep I went there. Have a docker image that has the packages already installed. This is pretty popular for reproducibility, but not great when there are frequent environment/package changes.

HackerFoo, 2019-04-09:
I would like to propose:

4. Use [Nix](https://nixos.org/nix/), which allows caching binaries as well as statically fixing dependencies to prevent surprises. This gives the stability of a monorepo without giving up modularity. It is portable to Linux and MacOS, but more importantly it forces all dependencies to be enumerated, which would help when making packages for other systems.

I agree that it may be too early, but I did some initial work on this: #563

hzeller, 2019-04-09:
I like solution (2) - have one project that bundles together all kinds of sub-projects. To do hermetic builds, a sprinkle of (3) (Docker), but _only_ as a basic development environment.

Reason why I think this is good
  - Easy to see which part is where and what the dependencies are. Right now (with conda) the dependencies are resolved 'automagically', but it makes it really hard to find the relevant pieces if you get started.
  - Easy to do local modifications or set up an own branch for local work. For instance, If I want to work on the vpr and I can do that right there in my tree and compile from there. Or point the submodule to a fork of mine.
   - Things are automatically hermetic: if we built everything from a superproject that checks out the most relevant sub-projects, it compiles everything consistently. No worry about whatever version is in conda, or it it works with my libc etc. That stuff is automatically taken care of.
   - This would mean that many less toplevel things that might be needed locally for building (e.g. gcc/clang, bison, libc or Python3) would come from the build host, which is the right thing.  
   - for build-farms, a Docker solution is good to supply a standard 'base image', e.g. it could build from a minimal Debian, Ubuntu, Arch image etc.; this image contains essentially just `build-essentials` and `cmake`, but none of the other packages (e.g. vpr) that are in-flux and are built from source.

(Personally, I find the current conda set-up very complicated and rather really have to build everything in place. I think it would make _development_ easier (_distribution_ might be easier with conda; but I am developing :) )

(If the worry is, that this might take time to build the tools - (Re-)compilation of c/c++ stuff essentially is free if one uses [`ccache`](https://ccache.dev/), which is anyway one of the first things a typical developer installs on their machines)