f4pga/f4pga-arch-defs #564: cmake -DUSE_CONDA=FALSE is broken

HackerFoo, 2019-04-09:
```
sf-shell-chrootenv:dusty@hank:~/src/symbiflow-arch-defs/build$ cmake -DUSE_CONDA=FALSE ..
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.7.2", minimum required is "3") 
CMake Error at make/env.cmake:314 (message):
  Could not find program progressbar2.
Call Stack (most recent call first):
  CMakeLists.txt:60 (add_conda_pip)
```

[make/env.cmake:314](https://github.com/SymbiFlow/symbiflow-arch-defs/blob/master/make/env.cmake#L314)

I've tried commenting out those errors, but then I run into trouble with lines like:

```
get_target_property_required(BLAH_TARGET env BLAH_TARGET)
```

and removing `_required` leads to a dead end where those things really are required.

elmsfu, 2019-04-10:
I think the logic [here](https://github.com/SymbiFlow/symbiflow-arch-defs/blob/master/make/env.cmake#L308-L318) is not quite right. If doesn't look at the `NO_EXE` flag. It should probably be checking if python can import those packages instead of `find_program` if `NO_EXE` is provided.

You can probably work around by setting environment variables before running `cmake`

mithro, 2019-04-10:
I think when using conda, the system is checking for progressbar2 being installed in the conda environment?

mithro, 2019-04-10:
`python -c "import progressbar2"; echo $?` could be the correct way to check the module is available?

litghost, 2020-07-14:
I believe this can be closed?

HackerFoo, 2020-07-15:
Yes, fixed with https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1521.

closed 2020-07-15 by HackerFoo