f4pga/f4pga-arch-defs #571: Rename `library` to `common/verilog`

mithro, 2019-04-10:
I think it makes more sense for the common verilog code under [`library`](https://github.com/SymbiFlow/symbiflow-arch-defs/tree/master/library) be moved to `common/verilog`.

We have XML stuff under [`common/xml`](https://github.com/SymbiFlow/symbiflow-arch-defs/tree/master/common/xml).

(It also might make sense to move `make` to `common/cmake` too?)

mithro, 2019-04-10:
@litghost @elmsfu Thoughts?

litghost, 2019-04-10:
I don't think moving everything under common is helpful, it adds hierarchy where one isn't really required.

litghost, 2019-04-10:
More specifically, does common/xml have anything to do with common/verilog?  No, then why put them in a tree together.

mithro, 2019-04-10:
@litghost library is just a generic name. It should have verilog in its name somewhere?

mithro, 2019-04-10:
It's also good to have a clean top level in the hierarchy. I would suggest we want to try and keep it to one per architecture and then a couple of others.

elmsfu, 2019-04-10:
I think `common/verilog` is better than `library`. But maybe @litghost has a point in that. Do we want `common` to be a bag of part?. Maybe `shared_verilog` or `verilog_lib` but we also don't want to promote everything to top level directory. `common/xml` should probably also be `xml_schemas`

mithro, 2019-04-10:
I was using common to mean "usable for all architectures".

litghost, 2019-04-10:
I do think `verilog_lib` is better than `common/verilog`.  I think a good example of something that is fine is `tests`.  I don't think `tests` should move under `common`.