f4pga/f4pga-arch-defs #572: Add SymbiYosys support for equivalence testing

litghost, 2019-04-10:
ZipCPU was kind enough to help debug the equivalence test failure in https://github.com/SymbiFlow/symbiflow-arch-defs/pull/570 and was able to prove the test case by using SymbiYosys, which is not currently supported in symbiflow-arch-defs.

litghost, 2019-04-11:
One downside of adding SymbiYosys support is the need to add an SMT2 engine.  While SymbiYosys + yosys can do proving via yosys-abc, it connect generate the failure case without an SMT2 engine (per https://github.com/YosysHQ/SymbiYosys/issues/40).  Options are:
- yices
- boolector
- z3
- mathsat
- cvc4

With the default as yices.  If we want to generate failure case, we'll need to bring in another binary :(

mithro, 2019-04-11:
* https://anaconda.org/asmeurer/z3

mithro, 2019-04-11:
Kinda surprised that the others are also not already packaged...

mithro, 2019-04-11:
CVC4 optionally links against the following libraries:
 * ABC                (https://bitbucket.org/alanmi/abc)
 * CaDiCaL            (https://github.com/arminbiere/cadical)
 * CryptoMiniSat      (https://github.com/msoos/cryptominisat)
 * LFSC checker       (https://github.com/CVC4/LFSC)
 *   http://www.ginac.de/CLN/
 *   https://github.com/timothy-king/glpk-cut-log